ArduPilot/ardupilot #2525: AC3.3: usb-connected fix for AUAV-X2 boards

rmackay9, 2015-07-07:
as discussed on the dev call, we need a workaround to detect the USB is connected for AUAV-X2 boards.

The plan is to add another check into the PX4 HAL so check that the usb-pin is high and that the USB device exists.  If either of these is false usb-connected will become false (false is safer because the vehicle will think it's flying), otherwise it will be true.

nickarsov, 2015-07-07:
Hi Randy,
Could this help?

"Suspended state:
The OTG_FS peripheral constantly monitors the USB activity. After counting 3 ms of USB
idleness, the early suspend interrupt (ESUSP bit in OTG_FS_GINTSTS) is issued, and
confirmed 3 ms later, if appropriate, by the suspend interrupt (USBSUSP bit in
OTG_FS_GINTSTS). The device suspend bit is then automatically set in the device status
register (SUSPSTS bit in OTG_FS_DSTS) and the OTG_FS enters the suspended state"

rmackay9, 2015-07-08:
Fixed by Tridge with this commit.
https://github.com/diydrones/ardupilot/commit/f4a9f9876fa2bb7803a838110580eae15ffb91ee

closed 2015-07-08 by rmackay9

nickarsov, 2015-07-08:
Hi guys,
Thank you very much!