ArduPilot/ardupilot #2528: Plane: Abstract control surfaces logic

fhedberg, 2015-07-07:
Abstract control surfaces logic (mixing, output, etc. mostly in attitude.cpp) and create an AP_ControlSurfaces library that can be shared between plane and copter (for hybrid vehicles).

lekston, 2015-07-31:
I do agree that this would be a great starting point for supporting hybrid VTOL vehicles. Lately, I was giving some thought on being able to run both the AC and Plane SW on a single PX4/Pixhawk board, i.e. having both apps in the same build and switch between them while in flight... In any case, many pieces of SW are common (AHRS, GPS support, Mavlink, Compass etc.), but some are quite different (Waypoint navigation, rc/motors support). One approach (as suggested above) is to have one abstract library to support both planes and copters. On the other hand it could be quite simpler to separate the common part into a single mutually used app running on NuttX and have two other apps (plane control/copter control) running in parallel. In this approach user would decide which aircraft mode is used by the VTOL... of course there still remains quite a lot of considerations related to the transition phase of VTOL craft (speeding up to convert to airplane mode and slowing down to convert to copter mode). FireFly is one example where two separate controllers are running each mode, it is a good place to start, but at the same time why need to carry two controllers if a single PIXHAWK would be able to support all required computations...

magicrub, 2015-07-31:
A long term goal is to run all playground in a single binary and switching
between them via param. That makes hybrid modes much easier to implement.

lekston, 2015-07-31:
Hm, is there anybody already working on this topic in the ArduPilot community? I would be quite interested in participating on this, but I would certainly appreciate guidance from somebody more experienced than myself...

magicrub, 2015-07-31:
Most of the developers have this in mind when making new commits. However,
it's a major task and would break *lots* of stuff if done all at once so
it's being done in safe tiny pieces where convenient. You're more then
welcome to help! Install mumble
<http://dev.ardupilot.com/wiki/ardupilot-mumble-server/> and come to a dev
call on Monday @5pm PST.

lekston, 2015-07-31:
Thanks! I will be glad to participate :) 
(it will be 2am CEST but I will do my best to stay awake)

magicrub, 2015-07-31:
Lets make sure Craig Elder puts a discussion item on the meeting agenda.

lekston, 2015-08-01:
Perfect, I will be there. Regards, Przemek.

magicrub, 2015-08-02:
If this is your first time on the mumble chat, please visit before the
meeting starts to work out any audio configuration issues.