ArduPilot/ardupilot #2529: Arming check doesn't check RC failsafe

squilter, 2015-07-07:
This issue was first identified at DroidPlanner/DroneKit-Android#151.  I was able to reproduce it using AC3.3rc7, `ARM_CHECK`=1

Turn off the RC, or disconnect the receiver from the pixhawk.
The GCS will report "RC Failsafe".
Sending the arming packet will still allow it to arm.

R-Lefebvre, 2015-07-07:
I'm not sure this is undesired.  We do attempt to support flight without a traditional RCTx.

squilter, 2015-07-07:
@R-Lefebvre My understanding was that this is the desired behaviour when `ARMING_CHECK`= -65 (Skip RC)

Pedals2Paddles, 2020-02-06:
Closing due no longer an issue. Copter will not arm by GCS (or anything else) if in RC Failsafe without disabling the arming check bit.

closed 2020-02-06 by Pedals2Paddles