ArduPilot/ardupilot #2532: Telemetry flow control (RTSCTS) automatic detection not reliable, especially with copter 3.3rc7

brad112358, 2015-07-08:
For some time (perhaps since around copter 3.2), I have occasionally had trouble with telemetry via radio on serial 1 not receiving any data at the GCS end in spite of the radio link being established.  The problem is only on startup, once it works, it works reliably for the entire flight.  Usually, restarting pixhawk (via reset or power) resolves the problem and the GCS connection starts working.  I recently updated to copter 3.3rc7 and the problem became much worse with most restarts not resulting in a usable connection to GCS.  The radio's LEDs show no traffic is being sent from the pixhawk side.

My radio is not wired for flow control (4 wire interface only) so, on a hunch, I changed BRD_SER1_RTSCTS from the default of 2 (automatic flow control) to 0 (no flow control).  Now the GCS connects reliably on every startup.  I took a look at PX4UARTDriver::_write_fd() and the logic looks reasonable, but perhaps there is some race with the UART or something which causes the check for blocked output to fail.

I will investigate further as time allows.  In the mean time I am reporting this with the workaround above in case others are seeing it.  If someone is willing to try to reproduce this issue in 3.3rc7 by temporarily pulling the RTS/CTS pins from the housing of the telemetry cable (if a 6 wire interface is used), I would appreciate it.  The RCTimer radios have only 4 wire interface (which is a shame).  Has anyone else noticed this behavior?

magicrub, 2015-07-08:
I may have seen this too but never investigated, I always just rebooted to get'r'done. If I start having a problem again I'll dive into it with flow-control in mind.

staroselskii, 2015-07-08:
@brad112358 @magicrub 

Is it possible that CTS and GND are shortened? This might force the bootloader mode. Once you get stuck you can verify it's not the issue. It's an easy check to perform, anyway.

magicrub, 2015-07-08:
Is the issue that CTS and GND are shorted or is it that CTS is low? A
simple reboot (usually) fixes is so there's no abnormal short happening.
I've seen this in 4 wire (no flow) and 6 wire (/w flow) connections to 3DR
900MHz modules.

brad112358, 2015-07-08:
@magicrub, If your problem was what I seem to be seeing, I wouldn't have expected you would see it with a 6 wire connection to the radio.  Are you sure you saw it on both 6 and 4 wire?

magicrub, 2015-07-09:
I'll check. Maybe that's why I hadn't seen it recently. However, if pixhawk
flow is of then the floor pins would clog up the radio Fifi because it's
not controlling the pin as it expects since the pin is being driven. I'll
check if that is the case by setting the flow in both modes

brad112358, 2015-07-09:
@magicrub,
If you connect a 6 pins, the radio will be able to send even if it is configured for flow control but the Pixhawk is not configured for flow control because CTS out is driven high from the Pixhawk even with flow control is not configured.  On the other hand, since a 4 wire connection does not connect CTS/RTS, configuring either end for flow control will cause a problem because the RTS input floats low.  Seems like a pull-up resistor on the CTS input would have made the heuristic used by the automatic flow control configuration unnecessary.  Are you able to reproduce the problem on 3.3rc7 with a 4 wire cable and the automatic flow setting?

magicrub, 2015-07-09:
Havnt tried yet. Also, I'm a plane guy so I'll be doing it on master
instead of a copper beta release.

brad112358, 2015-07-10:
I spent some time better characterizing the current behavior.  If desired, this testing can be done without even powering up the GCS radio by observing the LEDs on the radio connected to Pixhawk.  With no additional telemetry messages configured via SR1 parameters, which I believe leaves just keepalive messages, it takes almost 2 minutes for the automatic flow control code to turn flow control off and allow output without RTS/CTS connected.  This is the "working" behavior because the algorithm requires the large output buffer to fill first.  But sometimes, perhaps half of the time, the detection of flow control blockage fails completely and does not to recover from the missing wires resulting in no communication even after 10 minutes (as much patience as I had).  I believe it will never recover in this case.

I also examined the code and I believe what is happening is that bytes previously written are still in the buffer when flow control is turned on which sometimes causes _os_write_buffer_size to be calculated incorrectly because of a race with the interrupt level and also some strange code which seems to be a workaround for a bug I suspect has been fixed (or was a failed attempt to fix the race).  It is trivial and safe to fix the blocked forever behavior (I have a tested fix for this), but it should also be possible to reduce the time required to unblock as well.  I'll try to find time to work up a pull request this weekend.  If a developer decides to investigate before I do, please let me know so I don't waste my time.

magicrub, 2015-07-10:
Great test, thank you for investigating! It makes sense that once iy
switched modes that the buffer contents should be flushed. Is that the
solution you were implying?

As buffers get larger to support MAVLink data flash, I suggest we re-think
this feature.

This will effect me more soon with the data flash thing so I think I may
research it some.

brad112358, 2015-07-13:
I did work up a fix over the weekend.  It does allow the buffer to drain before the stuck in boot loader hack changes the baud rate and flow control back to configured.  That trivial change alone would fix the stuck forever bug, but I also completely reworked the strategy for flow control blocked detection to not require the buffer to fill completely before detecting we are stuck.  My fix reduces the maximum time to switch off flow control from almost 2 minutes to a second or two.  I will submit a pull request soon.  Hopefully after I figure out how to do a test build for vrbrain since that HAL needs the fix too.

I'm not sure if this fix should go into 3.3 or just the trunk.  I consider it well tested on Pixhawk, but I don't have any way to test the (identical) vrbrain changes.

magicrub, 2015-07-13:
Trunk

magicrub, 2015-07-13:
Master

magicrub, 2015-07-13:
The VRBrain UART driver is almost identical to the PX4. I'd like to review your fix in a PR first but generally I don't see any problem with applying your fix to both HALs while only being able to test it on one.

brad112358, 2015-07-14:
OK.  I've submitted the PR.  Let me know what you think!

brad112358, 2015-07-20:
Some other people seem to have encountered this issue as well.
http://ardupilot.com/forum/viewtopic.php?f=15&t=12724&start=10

brad112358, 2015-07-20:
And perhaps other projects are getting blamed...
https://github.com/DroidPlanner/Tower/issues/1487

brad112358, 2015-08-03:
Here is another case where others have encountered this bug and not realized what the problem was.
http://diydrones.com/xn/detail/705844:Comment:2062021
Note also that this report confirms my belief that bug apparently did not generally manifest itself in earlier releases.

brad112358, 2015-08-21:
I believe this bug is now resolved in the source and is available at least for copter as of the AC3.3rc9 test release.

closed 2015-08-21 by brad112358

AndKe, 2015-08-28:
please check last comments on https://github.com/DroidPlanner/Tower/issues/1487
- the issue is somehow different now, but not completely fixed