ArduPilot/ardupilot #2549: AC3.3: confirm take-off before GPS (both bad block & disconnected) flies ok

rmackay9, 2015-07-11:
Pomaroli on AC3.3 thread reports copter's roll/pitch is not good if taking off before GPS lock is attained.

btreeinfinity, 2015-07-17:
This one cost me $1400, grrr.

magicrub, 2015-07-17:
Sorry to hear about your crash, Charles. Don't forget that they call it
bleeding edge tech for a reason.

btreeinfinity, 2015-07-18:
I got over it, actually it went ape-**** and flew about 1/4 mile out, where a person grabbed it and ran off! So, I have been officially motivated to protect us all. I am working on a secured platform setup for the flight software. Blah blah blah.

rmackay9, 2015-07-18:
@btreeinfinity, If you have a log (which maybe you don't because it's gone with the thief), please feel free to post it on the AC3.3 beta testing thread so we can have a look.
AC3.2.1 flies fine with no GPS, AC3.3 has mixed reports on flying with no GPS - some say it's fine, one person said it was wobbly.

rmackay9, 2015-08-26:
We've had confirmation on the Copter-3.3 thread from two users that it flies ok without GPS so closing this issue.  If we get logs from people showing it doesn't then we can reopen.

closed 2015-08-26 by rmackay9