ArduPilot/ardupilot #2551: Plane: 3D accel calibration fails

codertux, 2015-07-12:
Hi guys,

Just upgraded tot 3.3.0 and I'm trying to calibrate the accelerometers. (Using APM Planner 2 on Mac OS X)

For some reason, the 3d calibration is failing. Here's what I get in the logs:
```
[Bix3:1] command received: 
[Bix3:1] Place vehicle level and press any key. 
[Bix3:1] command received: 
[Bix3:1] Initialising APM... 
[Bix3:1] Initialising APM... 
[Bix3:1] Initialising APM... 
[Bix3:1] Initialising APM... 
[Bix3:1] Initialising APM... 
[Bix3:1] Initialising APM... 
[Bix3:1] Calibration FAILED 
[Bix3:1] FAILURE: Failed CMD: 241 
```
The 1d calibration seems to work, but I still can't arm the plane.
```
[Bix3:1] command received: 
[Bix3:1] Initialising APM... 
[Bix3:1] SUCCESS: Executed CMD: 241 
```

Any ideas what's going on? I'll try to revert to the 3.2 series to try to rule out any hardware issues (this APM hasn't flown in a few months), but in the meantime maybe someone can help troubleshoot this on 3.3.

Thanks,
Andrei

staroselskii, 2015-07-12:
@chapman 

You can try with a newer GCS (i.e. [2.0.18](http://firmware.diydrones.com/Tools/APMPlanner/beta/?C=M;O=D)). I have definitively seen the issue at hand depend on the APM Planner's version.

magicrub, 2015-07-12:
This works fine on Mission Planner so this issue should be moved to https://github.com/diydrones/apm_planner/issues

closed 2015-07-12 by magicrub

codertux, 2015-07-13:
Thanks for the hint @staroselskii, I was able to perform the calibration using APM Planner 2.0.18.