ArduPilot/ardupilot #2552: AC 3.3 - Allow Serial5 port to be available for general use

utkinpol, 2015-07-13:
Hi, Issue relates to the restriction on available serial ports in a dual GPS setup with Serial1 used for telemetry, Serial 2,3,4 used for GPS, Teensy, minimOSD, etc. 
Integration with Storm32 controller requires a dedicated serial port and it would be helpful to have Serial5 port also available. 
On certain boards such as AUAV x2 Serial4 port does not exist at all and it limits functionality even more.
It would be very helpful to have Serial5 port available.

Ivosch, 2016-11-08:
Please plan this feature to the next release, I would appreciate it too.

peterbarker, 2017-12-19:
We can do this in master now.

closed 2017-12-19 by peterbarker

simjman, 2018-04-13:
Is it possible to get this working in Antenna Tracker?

amilcarlucas, 2018-04-13:
Did you compiled the antenna tracker source code from master branch and tested it ?
What exactly did not work ?

simjman, 2018-04-13:
Yes I compiled the antenna tracker source code as I understand that it pulls this change from the libraries. I am not familiar with C#, but decided to have a look anyway. Would be interested to learn more, but don't really have the time at the moment - already wasted a couple of days following the logic behind the mavlink side of things.

Anyway, as soon as sysid_thismav changes (in my case, to 2), I was no longer able to connect to the pixhawk. It would sit in an endless loop of "getting params". When I changed the aerial platform to 10 and the tracker back to 1, it would connect, but would keep cycling the connection between the aerial platform and tracker and wouldn't show the parameters for the aerial platform in mission planner. Hope that makes sense.

amilcarlucas, 2018-04-13:
The source code for the tracker is written in C++, not C#. that one needs to be build from source.

Make sure you use the latest Mission Planner 1.3.56 or something like that. MP is written in C#. But there is no need to recompile it from source AFAIK.

simjman, 2018-04-13:
Ok no worries. My misunderstanding of the different languages. I'm not familiar with C++ either. I built it with source using the px4 console and I've also messed around in eclipse. 

I should note that I have previously had the tracker working on v1.0.0 firmware, but now need the extra serial port.

simjman, 2018-04-13:
Also, I am using the latest version of mp

rmackay9, 2018-04-14:
No need to even compile it yourself actually.  Just go to the MP's Install Firmware screen and press Ctrl-Q to get the bleeding edge developer version.

simjman, 2018-04-14:
Ok good to know, will give it a crack now.

simjman, 2018-04-18:
Ok so it does give the dev version for tracker even though the label doesnt update, however this gave the exact same problem I experienced from when I was building the firmware myself. The tracker gets hung up as soon as the vehicle connects and the sysid for the tracker changes (in this case to 2).

I tried changing the sysid of the vehicle and setting the tracker to 1, but this caused the tracker and vehicle to constantly cycle the connection. I understand that this isn't necessarily your issue, but changing the sysid of the vehicle from 1 also stopped the teensy on the tracker from sending telemetry to the taranis - this may be a simple change to the teensy code, but haven't looked into it.

Also, as a side note (and yes I understand this is more for the support forums or app development) does anyone know if it is possible for the tower app to work properly with the antenna tracker?

Appreciate your time and help.

simjman, 2018-05-01:
Is there going to be any traction on this or do I need to create an issue?