ArduPilot/ardupilot #2553: Plane: automatic parachute release

naktinis, 2015-07-13:
Add the automatic parachute release feature to Plane. Something similar to what is currently implemented for Copter: http://copter.ardupilot.com/wiki/common-optional-hardware/parachute/

As I understand, this is the essential bit of code: https://github.com/diydrones/ardupilot/blob/master/ArduCopter/crash_check.cpp#L65

rmackay9, 2015-07-13:
Duplicate of this existing issue: https://github.com/diydrones/ardupilot/issues/2359

naktinis, 2015-07-13:
I saw #2359, but its description said "enable deployment from a mission command", so I thought it is just about that and not about an automatic triggering.

naktinis, 2015-11-04:
This was marked as a duplicate of #2359, which was closed, but the functionality described in this issue has not been implemented there. As, from what I understand, automatic parachute deployment is a desired feature (it works in Copter), could someone reopen this ticket?

With some guidance I might try and implement basic functionality (say, trigger parachute when not landing/taking off and height falls between two given configuration parameters). But as I've not yet contributed to Ardupilot, I could use a couple of pointers (adding parameters, checking altitude in some crash check loop).

bchristal, 2015-11-20:
Just adding a +1 to implement, at least at a manual or mission triggered event if that's less problematic than determining an out of control airplane. Would be great if "Chute" could be used as a landing/panic method. Basically just activate a servo AND kill throttle.

naktinis, 2015-11-23:
@bchristal I think 5d1d54a8cb, d20b86b95e, and #2359 are about manual parachute triggering (that kills the throttle as well) via a specified channel. Would this work for you? As I understand, we want this basic functionality tested before implementing automated deployment.

naktinis, 2016-02-23:
I'm considering implementing a very basic automatic parachute deployment. The idea would be to automatically trigger the parachute if the plane is cruising in AUTO (is_flying, not landing, not taking off) and it is off from desired altitude by the specified distance for more than two seconds, but still above CHUTE_ALT_MIN. Of course, one would need to explicitly turn the automatic deployment on for it to work.

Here's a rough outline for this (I've never contributed to Ardupilot, so this could have many flaws).

New parameters:
CHUTE_AUTO_ENABLED (turn automatic deployment on/off, default: off)
CHUTE_AUTO_ERROR_ALT (deviation from desired altitude at which to deploy the parachute automaticallys in meters)

Code changes:
- **parachute.cpp**
 - add `Plane::parachute_auto_check()`, which:
   - checks if mode is AUTO, flying, not landing, not taking off, parachute enabled, CHUTE_AUTO_ENABLED enabled, parachute not released, above CHUTE_ALT_MIN, otherwise returns
    - if altitude_error_cm / 100 > CHUTE_AUTO_ERROR_ALT for more than 2 (may be configurable) seconds, then releases parachute and turn off throttle
- **ArduPlane.cpp**
 - SCHED_TASK(parachute_auto_check, 10, 500)

Questions/notes:
- Does the general idea sound reasonable and would it work as a very basic initial automatic parachute deployment logic?
- Naming conventions – do the parameter and method names look ok?
- Actually I'd rather implement this in just `parachute_check` (like copter does in crash_check.cpp), but for some reason current `parachute_check` (parachute.cpp) serves as a helper for manual release – just calls the lower level function and doesn't do much else
- Any suggestions for implementation details
- How should this relate to failsafe logic (it does not in Copter parachute deployment, as far as I know)?

magicrub, 2016-02-23:
Rim,

Chutes are good! However, it's been implemented already via auto mission
item
<http://plane.ardupilot.com/wiki/common-mavlink-mission-command-messages-mav_cmd/#mav_cmd_do_parachute>.
Please take a look at libraries/AP_Parachute
<https://github.com/diydrones/ardupilot/tree/master/libraries/AP_Parachute> and
either close this or add to what is already implemented.

naktinis, 2016-02-23:
@magicrub, I believe MAV_CMD_DO_PARACHUTE is about releasing the parachute as a planned part of a mission and I'm talking about automatic emergency detection (e.g. plane unexpectedly starts losing altitude during mission; may be an engine fault, may be a broken wing, may be stalling etc.) and then triggering the parachute as a result.

I know about `libraries/AP_Parachute`, but I think it is about mechanically triggering the servo, not about detecting when the plane is about to crash (which I'm trying to implement) and which looks like more suitable for `ArduPlane/parachute.cpp`.

magicrub, 2016-02-23:
In that case, please take a look at the Failsafe parachute features by
searching "chute" on the params page
<http://plane.ardupilot.com/wiki/arduplane-parameters/>. You'll find
FS_LONG_ACTN
<http://plane.ardupilot.com/wiki/arduplane-parameters/#long_failsafe_action_arduplanefs_long_actn>
== 3 is "Deploy Parachute". However, there's no free-fall detection, only
altitude, so that would need to be added.. or whatever logic you propose.

naktinis, 2016-02-23:
@magicrub I don't see any failsafe triggered by altitude error in AUTO mode, so what exactly do you mean by "only altitude" detection being available for FS_LONG_ACTN?

My proposal above is inspired by what is done for ArduCopter here: https://github.com/diydrones/ardupilot/blob/master/ArduCopter/crash_check.cpp, compare it to https://github.com/diydrones/ardupilot/blob/master/ArduPlane/parachute.cpp. I see obvious parallels: both have `parachute_check()`, `parachute_release()`, and `parachute_manual_release()` methods.

What I'm proposing in my comment above is basically implementing a simple (for now) ArduPlane equivalent of ArduCopter's `crash_check()`.

naktinis, 2016-02-25:
@tridge, @rmackay9 since you've worked on the parachute code, could you please have a brief look at my proposal above?

naktinis, 2016-02-26:
Started implementing it here: https://github.com/laserpas/ardupilot/commit/ba50d7847b609a98a051aa53dbdf5c366bf68e99

The idea is essentially as described above: when flying in AUTO (not landing, not taking off, and auto parachute release enabled) look for large altitude errors and if the error is too great for 1 second (subject to revision), then release parachute without user's intervention.

Tested using SITL (with `param set SIM_ENGINE_MUL 0`).

Comments and ideas welcome.

Naterater, 2018-12-28:
The failsafe chute release action is complete. http://ardupilot.org/plane/docs/parameters.html#fs-long-actn-long-failsafe-action

If you want more items to trigger failsafe logic, then that should probably be raised as a separate failsafe issue - the chute logic is complete.

closed 2018-12-28 by Naterater