ArduPilot/ardupilot #2559: AC3.3: enable EKF check only after EKF position estimate has been initialised

rmackay9, 2015-07-14:
This should resolve the issue where the LEDs flash yellow-red if the GPS is disabled (GPS_TYPE=0) intentionally and the pilot is only flying in non-GPS modes.

andyp1per, 2015-07-14:
IIRC I still get the blue light when I am in a non-GPS mode and don't have a lock (not actually tried flying). Is this a different scenario?

rmackay9, 2015-07-14:
During a bench test this is certainly the case, but in a real copter I think it flashes yellow+red.  There's a report on the AC3.3 beta testing thread from another user as well.

andyp1per, 2015-07-14:
I never do benchtests :) I'll have another go to check when I have some time.

rmackay9, 2015-07-16:
This should be resolved with this commit which will go out with AC3.3-rc8
https://github.com/diydrones/ardupilot/commit/277e3d86752da103a4d3a8ce1061664ab3d20bf6

closed 2015-07-16 by rmackay9