ArduPilot/ardupilot #2568: Copter: loses altitude if pilot commands full lean in AltHold, PosHold, Loiter

rmackay9, 2015-07-18:
The lean angles command by the pilot in AltHold and PosHold take priority over the altitude hold.  This request (received from 3DR-solo and at least one other user) is to prioritise altitude over lean angle.  This means the pilot's input would not be respected but the vehicle would maintain altitude.

proficnc, 2015-07-18:
Would this be a parameter? Could the old behaviour be reverted by choice?

rmackay9, 2015-07-18:
We weren't planning to make it a parameter.  I think generally people don't want it to lose altitude .. but the question becomes whether the patch from Leonard and I and the bobbing that can result is acceptable or if we need to take a very different approach.

WickedShell, 2015-07-20:
I had always assumed AltHold/PosHold/Loiter always enforced the altitude over anything else. Given the names and descriptions that's the behavior I'd rather see as a user.

roque-canales, 2015-08-04:
Effectively, I recently experienced with overloaded drone, that AC3.3 no prioritize altitude, then that's what I was expecting.

It is very important that the drone maintain its altitude foremost, even if it must slow (reduce lean angle), not achieving its desired speed during auto mission or loiter (when max_speed set exceeds drone capability  or occasionally when wind is too strong for the drone), or if the pilot request to mutch lean angle in  althold.

johnbrownlow, 2015-08-14:
I agree holding altitude should be prioritized over lean angle. I encountered this today with a CX-20 upgraded to 3.2.1 and with the max lean angle set to 35 degrees. In PosHold, as soon as gave it full forward elevator it would lose altitude rapidly, very dangerous and not the expected behaviour from my point of view.

magicrub, 2015-08-14:
Send obvious to me that altitude should be prioritized in these modes. I've
suffered from this too.

johnbrownlow, 2015-08-14:
Where can I find the patch you mentioned upthread?

rmackay9, 2015-08-14:
The changes to prioritise altitude over lean-angle are here: https://github.com/rmackay9/rmackay9-ardupilot/commits/leo-thr3
Master has moved on a bit so there will be some merge issues when rebasing these patches on master.  I do intend to do that at some point but can't promise when.  I also can't promise that you'll like the results - it leads to some bobbing in pitch as it backs off the lean-angles.

To keep expectations reasonable, let's remember that of course there's no way to fix a vehicle power issue in software.  So setting the max lean angle to something far beyond what the vehicle is capable of is not useful.  The benefit of this patch is that for cases where the vehicle hits a headwind and needs additional lift tempoarily, it backs off on the lean angle.

crimsondr, 2015-08-14:
I actually do not like this proposed change.  The bobbing side effect won't be that great.  I also use the current behavior to descend quickly while avoiding VRS.  It would be great if this new fix could be enabled/disabled via parameters.

rmackay9, 2015-09-24:
This is in master for AltHold.  So people should give it a try and report back if they see problems.

srobi, 2016-03-06:
Is it part of and release/rc yet? I will start flying soon and happy to test as I needed this change.

lthall, 2018-01-11:
This has been done.

closed 2018-01-11 by lthall