ArduPilot/ardupilot #2571: Copter/Plane Request: CAM_TRIGG_MAX

iskess, 2015-07-20:
A new parameter called CAM_TRIGG_MAX that defines the maximum firing rate of the camera. While firing using CAM_TRIGG_DIST, the ground speed will not cause this rate to be exceeded. In this case the overlap of the images may start to suffer, but that is preferable to the uncertainty of dropped frames. 
Ideally a GCS alert could notify the pilot that the camera's capabilities are being exceeded.

WickedShell, 2015-07-20:
If you have exceeded the CAM_TRIGG_MAX and want to take a photo and defer it, do you then take the photo as soon as you can according to the time, or do you still wait for the next time the distance comes up?

iskess, 2015-07-20:
No photos get deferred. If the camera's max fire rate is 1 shot/sec and a tailwind causes CAM_TRIGG_DIST to try to fire every .9 secs, the triggering command will not occur more frequently than 1/sec due to this new parameter. 

Currently if the camera's max rate is exceeded by the AP's commands the camera will drop a frame somewhere. This unpredictable behavior causes a great deal of confusion in the geotagging process amount other issues.

WickedShell, 2015-07-20:
Let me restate it I guess, if we take the maximum rate of 1 photo a second, and we've traveled the distance after 0.9 seconds, don't take the photo, but as soon as we go over 1 second (so 0.1 seconds later) take the photo. This would reduce the amount of photos lost as much as possible. Is that what you're looking for?

iskess, 2015-07-20:
Yes, you just stated it differently than i was thinking. My thinking was that the code would switch from distance based to time based to maintain 1 sec intervals. You are approaching it by adjusting the distance based logic. I'm not a coder, so whichever makes more sense to the Devs. The results should be the same.

ThorstenDP, 2015-07-20:
Hi all,
I think this can be easily done when planning a mission with a GCS - especially when you frequently change the camera. I opened an issue for Tower (https://github.com/DroidPlanner/Tower/issues/325) and made a mock-up of some more feature rich mission planning tool
(https://github.com/DroidPlanner/Tower/issues/325#issuecomment-27289331) quite a while back.

WickedShell, 2015-07-20:
The primary purpose of showing it in the ground station is to figure out a flight plan that won't exceed the trigger rate. The proposed change in the code (which makes sense to me, and I should get to fairly quickly) will only limit how much overlap is lost if you are flying to fast. It still doesn't solve the fundamental problem which is simply a failure in mission planning (and can indeed be figured out well in the ground station).

iskess, 2015-07-20:
The mission planning software doesn't account for wind. If there were an entry for prevailing wind velocity, it would be easier to account for ground speed changes and camera speed limits, asymmetrical turn overshoot distances, increased battery consumption due to large crab angles, and flight time. But thats beyond the scope of this issue entry. 
Thank you for giving this your attention, I'm happy to hear you agree.

Jman841, 2015-07-20:
This is a great idea. I have had issues with this in the past where the logging gets off and then the pics can't be geotagged.

Could we also add a test function to test the max trigger rate of the camera? So if we set this parameter to a max of 1 picture per second, we can run a test for say 5 or 10 seconds of it being triggered at 1 picture per second to ensure it works well?

WickedShell, 2015-07-20:
As far as geotagging goes you can use this http://swiftradioplanes.com/swifttag/ to geotag photos as it doesn't care if you have more CAM's then photos, its explicitly designed to handle that. (It needs an update for Plane 3.3 and Copter 3.3RC's which should go out today).

The test function doesn't really correspond well with how I was going to implement it. It would require more state, and doesn't seem to offer anything. If you simply want to test on the ground you can use a script in mission planner to trigger the camera at a certain interval or trigger it from the GCS manually at the desired interval. I can't see putting a test function on the autopilot. To determine the rate at which photos can be taken you can easily test that manually or with the GCS.

iskess, 2015-07-20:
Please see issue #2410 request for an intervalometer. If this were implemented it could also function as a test. 
I don't code, but the Arduino "blink" script was simple enough and serves as a great bench test for camera speed.

WickedShell, 2015-07-20:
I just can't see what #2410 provides that is useful. I can't come up with a scenario in which it's not inferior for mapping in every way.

iskess, 2015-07-20:
Yes I struggled with the same question which @robustini didn't really answer. 
The best I could come up with is of you wanted your camera to fire at its absolute maximum rate all the time to get the most photos possible. But my request could handle that in most circumstances by choosing a CAM_TRIGG_DIST of 1, and setting CAM_TRIGG_MAX to the camera's limit. However if a slow aircraft is using a very fast camera we might need a CAM_TRIGG_DIST below 1 for this to work.
Perhaps it could be useful for something other than mapping, like flashing a strobe at a set rate?
Or perhaps you are taking surveillance photos and you are hovering or loitering over ROI? Distance based triggering would never work. I'm sure there are more uses that I haven't thought of, and the code is so simple.

Jman841, 2015-07-20:
How do you run a script in mission planner to trigger the camera X times in Y time frame?

WickedShell, 2015-07-21:
Modified from Mission Planner's Example 2 the following will command pictures continuously every 1.5 seconds (just change the delay time). If you really need it I can modify it so it does x number of pictures, but I would say just stop the script once you're happy with the camera's performance. (It may not be the most optimized script, I hacked apart example 2 really quickly when I needed it)

```python
import clr
import MissionPlanner
clr.AddReference("MAVLink")
from System import Byte
import MAVLink
from MAVLink import mavlink_command_long_t

import MAVLink

print 'Start Script'

i = 0

while True:
  print 'Picture ' + str(i)
  i = i + 1
  MAV.doCommand(MAVLink.MAV_CMD.DO_DIGICAM_CONTROL, 0, 0, 0, 0, 0, 0, 0);

  MAV.doCommand(MAVLink.MAV_CMD.DO_DIGICAM_CONTROL, MAVLink.MAV_MOUNT_MODE.NEUTRAL.value__, 0, 0, 0, 0, 0, 0);

  MAV.setDigicamControl(True);
  Script.Sleep(1500)
```

iskess, 2016-03-07:
Was this ever implemented? I look through the parameters list on the wiki shows CAM_MIN_INTERVAL, but I don't see it in mission planner. I also see CAM_MAX_ROLL on the wiki param page. 
I can't find any reference to either parameter in the code.

closed 2016-03-15 by iskess

iskess, 2016-03-15:
I see CAM_MIN_INTERVAL was created to fulfill this issue. I haven't tested yet, but I'm closing it with the assumption it works as advertised. I don't know who did the PR, I don't see a record of it. Whoever it was, thank you very much.