ArduPilot/ardupilot #2572: AP_GPS: Log which GPS is being used

WickedShell, 2015-07-20:
In a dual GPS scenario there is no indication of which GPS is being used to provide position. This should at least be logged, if not reported to the GCS.

iskess, 2015-07-20:
Please see https://github.com/diydrones/MissionPlanner/issues/492

rmackay9, 2015-07-20:
@WickedShell, totally agree that we should log this in the dataflash.

magicrub, 2015-07-21:
+1, we should log this for sure! Doesn't need to be fast, 1hz is plenty. There's logic to inhibit how often we are allowed to switch.

WickedShell, 2015-08-05:
Logging aspect was covered in 5776620

We still lack a mechanism to report which GPS is the primary one to the GCS while in flight. Requires a MAVLink change.

WickedShell, 2017-06-09:
I'm no longer convinced a MAVLink change is all that useful, as we don't have anyway to report the synthetic GPS to the ground station. We either need a generic (with instance number) GPS reporting message to which we could add the flag (I'd prefer it on the one hand as it would give a uniform message for all GPS instances, but given that it would be yet another way to send GPS data I'm not excited),

Given the current constraints I'm wondering if a statustext would be sufficent everytime we swap what the primary instance is? Yes it could be dropped, but thats the same level of protection we give you for EKF core swapping, and it would be a more apparent notification to most users. Any thoughts on this approach?

OXINARF, 2017-06-11:
I'd prefer the new messages to using another statustext (we use too much of that, which make GCSs do all kind of hacks), but I do understand your issue with that approach.

WickedShell, 2017-06-11:
I had opened a MAVLink PR https://github.com/mavlink/mavlink/issues/496 hasn't gone very far, but I also haven't been pushing it as any kind of priority, and would need to revaluate what the proposed messages are :)

Statustexts were meant simply as an idea for how to do a best effort on this at the moment without having to create a new MAVLink message. :) Will go back to letting it sit.

peterbarker, 2021-07-30:
This story has gotten a lot more complicated since the introduction of sensor affinity in NavEKF3.

Which GPS being used is now no longer an attribute of the GPS library, it now also depends on which core the NavEKF3 is using - and if you've fallen back to DCM.

That makes the title of this issue actually incorrect as the GPS library has no idea which sensor is actually being used (and for what, given our sensor selection code...)

How should we proceed with this?  A statustext emit in AP_GPS many years ago may have been useful at the time, but now would be potentially misleading.