ArduPilot/ardupilot #2582: Copter Request: Set target pitch angle rather than WPNAV_SPEED

iskess, 2015-07-21:
I'm still rather new to Copters and approaching things with a Plane mentality so forgive me if I'm off base here. 
It doesn't seem to make sense to be mapping at a constant ground speed now that we have CAM_TRIGG_DIST. If I have WPNAV_SPEED set to 8m/s and the wind is blowing at 5m/s, why must the copter slow its airspeed down to 3m/s on the downwind leg. The biggest limitation with copters is endurance, so we should be flying at the most efficient airspeed, or in this case pitch angle. 
What is the most efficient pitch angle on a copter? 30 degrees maybe? Depends on drag too I'm sure. Whatever it is, I want to fly that angle during my missions and take advantage of that tailwind which I just paid dearly for on the upwind leg!
If anyone is worried about overspending the camera, then see

iskess, 2017-01-13:
Randy, you added some labels and didn't shoot down this idea so I remain hopeful. 
It's been about a year so I thought I'd check in to see if it's on the roadmap.

R-Lefebvre, 2017-03-31:
IMO, this would be quite sub-optimal.  This would reduce the system performance below where it's currently at.

A better solution to the root problem you are targeting, is to have airspeed estimate in ArduCopter, and fly at a target airspeed.

rmackay9, 2017-03-31:
It's likely that for some applications we need ground speed (i.e. when doing mapping) but in others airspeed would be better.   So perhaps a bit like how we accept different altitude types (terrain, relative, absolute) our mission commands should accept air speed or ground speed.  It will be tricky of course because our waypoints are locked to the ground (i.e. lat/lon) so for commands where we are stopping we will need to transition from airspeed to ground speed.