ArduPilot/ardupilot #2584: Copter: mission_set_current can't restart a mission

squilter, 2015-07-22:
The command mission_set_current allows a gcs to tell the copter to skip a waypoint or to rewind a mission.  Reading the comments [here](https://github.com/diydrones/ardupilot/blob/master/libraries/AP_Mission/AP_Mission.cpp#L312), it seems that the intended  behaviour is that the copter restarts a mission (after the mission is complete) when sent mission_set_current(0).  So I tried sending it after the copter had reached the last waypoint on the mission, and the copter said "going for waypoint 1".  But it just sat there, and wouldn't restart the mission.

Randy says:
"Ok, so the issue there probably is that the _flags.state can be != MISSION_RUNNING either when the vehicle’s not in the Auto flight mode or when the mission has completed.  We need to find a way to deal with the special case of the mission being complete but the vehicle is still in auto mode… just in those cases we want the state to change from mission-complete to mission-running."

hamishwillee, 2015-07-23:
Just FYI, it is possible to restart the mission by toggling the mode back to guided and then to AUTO again. This will then proceed to the next waypoint. Not a particularly good solution. Would prefer that the MISSION_START command was able to start a stalled mission.

KiwiHC16, 2015-08-06:
If i put resume() in "AP_Mission.cpp" line 356 of 1413 --25%-- col 3-10 then it works.
----------------------------
// if we got this far then the mission can safely be "resumed" from the specified index so we set the state to "stopped"
        _flags.state = MISSION_STOPPED;
        resume();
-----------------------------
Question: how to send a resume command through mavlink ? or should we resume automatically (probably not the best idea).

The resume mission button from Mission Planner, send few requests and not only a resume. I think it send arm, take off, guided, mission_set_current and then auto.

KiwiHC16, 2015-08-25:
In order to be able to send a message to call resume() I did the following.
s test: in file GCS_Mavlink.cpp in function GCS_MAVLINK::handleMessage changed the 
------------------------
   case MAV_CMD_NAV_LOITER_UNLIM:
            if (copter.set_mode(LOITER)) {
                result = MAV_RESULT_ACCEPTED;
            }
            break;
------------------------
by
------------------------
   case MAV_CMD_NAV_LOITER_UNLIM:
        	copter.mission.resume();
            break;
------------------------

At the end of the mission, last command is a WP, the copter Loiter and wait.
From Mission Planner Action Tab, Loiter_UNLIM, Do Action.
It's triggering the function resume() above and then copter start its mission again.

Should we add this function into Mission Planner (Resume Mission - Do Action) and the associated modification into GCS_Mavlink ?

KiwiHC16, 2015-08-25:
Even better solution: In Mission Planner - Actions - Mission_Start - Do Action button
And one row added with resume()) into GCS_Mavlink.cpp file:
.........................
        case MAV_CMD_MISSION_START:
            if (copter.motors.armed() && copter.set_mode(AUTO)) {
                copter.set_auto_armed(true);
                copter.mission.resume();
                result = MAV_RESULT_ACCEPTED;
            }
            break;
.........................
I think it's a good solution. After a mission, the copter wait. You can do a SET_CURRENT to WP 5 for example and then when you are ready: Mission_Start - Do Action (there is even a warning message with MP).
If anybody is against it, I will pull for a the modification.

hamishwillee, 2015-08-25:
I think it looks fine, but up to @rmackay9

magicrub, 2015-08-25:
Why not use copter.mission.start_or_resume()?

KiwiHC16, 2015-08-26:
Looks even better. Just tested it and works fine.
Any other recommandation ?
I'm going to propose a PR to @rmackay9

KiwiHC16, 2015-08-28:
I created the pull request 
https://github.com/diydrones/ardupilot/pull/2775

squilter, 2015-09-16:
Fixed with #2775

closed 2015-09-16 by squilter