ArduPilot/ardupilot #2587: LIDAR LITE calibration

Jman841, 2015-07-22:
Currently there is no documentation on the proper way to calibrate the LIDAR Lite when using the PWM connection.

On both a Quadcopter and Plane running the pixhawk with LIDAR Lite, when measuring the distance from over 1 meter and comparing it to the values the LIDAR states, the LIDAR states the value being about 20-30% more than the actual measurement is.

I assume as the distance gets larger, the offset is even worse.

I can adjust this by changing the Volts/Meter value from 1 to .9 or .8, however, I am not sure if this is the correct way to go about it or why the default value of 1 is so far off.

Thank you

magicrub, 2015-07-22:
There is a range finder ground offset value intended to account for landing
gear. I believe the default value is 20cm.

Jman841, 2015-07-22:
The issue is not what it thinks landed altitude is, the issue is the accuracy is off by 20-30% and the problem persists at longer distances. We tested about 10 meters and it was reading around 12. I think the voltage/meter value of 1 is not perfectly correct.

When I initially set it up via the I2C port the accuracy was dead on. Via the PWM installation the accuracy is much less precise.

Thank you for the help and comments, it's always appreciated!

h3ct0r, 2016-11-05:
@Jman841 I have better results using 0.8 in the scale also! Im using a Lidar Lite v1 from PWM.

gmorph, 2016-11-07:
I believe it will vary slightly from device to device which is why the Volts/Meter setting is there.  So your doing the correct thing by changing it to suit your LIDAR.  The PWM is of course an analog conversion subject to EMF noise and conversion issues whereas your I2C was a digital value which didn't have those problems (but it has other problems).
Thanks, Grant.

magicrub, 2016-11-07:
I've used it at 1 and it worked for me. Those units were a good value, no necessarily good.

h3ct0r, 2016-11-07:
@gmorph Maybe specifying how to tweak this parameter will help other people. I have open an issue on the wiki https://github.com/ArduPilot/ardupilot_wiki/issues/593 to help improve the docs.

If this a parameter that is not deterministic, showing other ranges of values, or where to tweak is going to be very helpful to novices.

magicrub, 2016-11-07:
fixed via wiki commit https://github.com/ArduPilot/ardupilot_wiki/commit/f7d79240a568f2aa9d67d784af793c0b95ab84ae

closed 2016-11-07 by magicrub

gmorph, 2016-11-07:
Yeah I've added more info to the Wiki under the Testing section at the bottom.

Thanks, Grant.