ArduPilot/ardupilot #2588: Lidar Lite altitude bump during landing

Jman841, 2015-07-22:
For Arduplane, I am using a LIDAR Lite to help with landing since the Baro Drift over an hour can be extreme and making automatic landings unsafe.

However, we have a few small tree's about 8 meters high before our runway and when using the Lidar Lite for landing it detects the tree's for a short time and then pitches up sharply on the plane which almost causes a stall, then when it pitches back down the speed is too high and it always overshoots the landing point much further than it should.

Could there be a safe way to ignore sudden large changes in altitude on the land sequence from the range finder?

Maybe have a 1 second delay during landing sequence of a sudden altitude change where the plane does not react, but, if it goes over that time period it assumes it is not just a single obstacle and then reacts? Or maybe it just goes to level flight for that 1 second to determine if it is just a single object or if it is in fact the ground and it needs to pitch up?

Any other idea's are most certainly welcome, however, the current method of extreme pitch up from a single tree before the runway is definitely not helping for a good landing pattern.

magicrub, 2015-07-22:
I have been working on this off and on. There will be a solution soon.

Jman841, 2015-07-22:
Excellent,

Thank you

magicrub, 2015-07-31:
I've created an overly fancy tree detection thing but it's too complicated and ended up needing to be tweaked for every different situation, couldn't get it to be generic enough. Meanwhile, I had an idea of simply low-passing the step input but it already is here, but there's a step input that can screw things up. I'm thinking simply removing that may help. Always low pass all the time.
https://github.com/diydrones/ardupilot/blob/8f0e7e0693405d30d05d1ab67d7ae855f1a561ec/ArduPlane/altitude.cpp#L591

if you don't get a new value then don't update the correct but still filter so all input corrections are smoothed. That allows the navigation controller to react to slower moving pitch inputs rather than step inputs which really screw things up during landing

magicrub, 2016-01-05:
@tridge and I have discussed a new slope calculation solution. I have a wip branch but needs a little more work and testing. Should be done soon.

magicrub, 2016-06-06:
this has been implemented with two params: [LAND_SLOPE_RCALC](http://ardupilot.org/plane/docs/parameters.html#land-slope-rcalc-landing-slope-re-calc-threshold) and [LAND_ABORT_DEG](http://ardupilot.org/plane/docs/parameters.html#land-abort-deg-landing-auto-abort-slope-threshold)

closed 2016-06-06 by magicrub