ArduPilot/ardupilot #2591: UBX messages messed up with multiple UBLOX GPS

andyp1per, 2015-07-22:
This is because there is a single message type which both GPS' end up logging to. I have a fix

WickedShell, 2015-07-22:
There is a instance field in the UBX1 i thought? We need to propegate it to
the other GPS messages. I really don't want to make more GPS 1 and 2
messages. Makes more sense to only have 1

andyp1per, 2015-07-22:
How do you graph them in GCS? This is inconsistent with MAG, MAG2, IMU, IMU2, GPS, GPS2 - so disagree unless you are going to change everything.
Also GPS and GPS2 have different fields - that's also highly confusing.

WickedShell, 2015-07-22:
Okay that's not a bad point. It still feels more like a tooling issue to
me, but additional messages doesn't cost us much. Just a bit of runtime
overhead, and its faster/easier to fix it in what we log you're right.

I've been using vim pretty heavily to edit logs before graphing them, as
it's far faster then any of the graph tools we currently have.

andyp1per, 2015-07-22:
I think for most users the GCS' we have are convenient enough for graphing - and most people would rather be out flying :)

WickedShell, 2015-07-23:
Your totally right. It'd be nice if the gcs could handle it but until then
your right.

WickedShell, 2015-09-24:
addressed with #2826

closed 2015-09-24 by WickedShell