ArduPilot/ardupilot #2594: Rover: NAV_WAYPOINT param1 time to stop in seconds isn't working

gmorph, 2015-07-23:
See this post:
http://diydrones.com/group/ardurover-user-group/forum/topic/show?id=705844%3ATopic%3A2053446&xg_source=msg
I've replicated the issue in SITL.

Thanks, Grant.

earsneyes, 2015-07-23:
I remember someone wanted to use this in a competition and wanted to hand over control while stopped to an companion computer running vision stuff.

Maybe we should put that on the list either ?
For example "wait for x or until companion computer is signaling go on".

Cheers
Linus

rmackay9, 2015-07-23:
@earsneyes, I wonder if I was that person.  I needed to do that for the red balloon popper at Sparkfun and so we implemented a new navigation command called MAV_NAV_GUIDED_ENABLE and a new "Do" command called DO_GUIDED_LIMITS.

For copter at least, normally Guided is the flight mode where the GCS or companion computer can control the vehicle but in this case we needed the companion computer to control the vehicle for just some portions of a mission so this new "NAV_GUIDED" mission command essentially temporarily ran the equivalent of the "Guided" flight mode within "Auto".

earsneyes, 2015-07-23:
@rmackay9 I think it was a Rover user request but good point about your balloon popper. 

I do not need it urgently, just wanted to add that this feature was requested in the past.

I might read through the balloon popper code to see if i can understand it. 
I dont think i am able to port that feature over to rover yet.

@gmorph Do you think we could do a hangout regarding rover in the near future if time permits ?

gmorph, 2015-07-27:
Happy to do a hangout anytime - you might have trouble stopping me from talking ;-)
As for the future of Rover these are the items on my TODO list (in priority order):
Kick butt in T3 (this is proving hard to arrange)
Arming
Enable changing to reverse at a waypoint - so still going to the next waypoint - just in reverse
Geofencing
Autotune
Breaking using a discrete break output channel to dedicated breaks on an actuator
Waypoint converstion to INT in MAVProxy (big task)

In amongst that list is bug fixing, wiki, doco, Dronecode, other vehicle work and doing anything Tridge requires of me.  All good fun!

Thanks, Grant.

gmorph, 2015-07-28:
Fixed with this PR
https://github.com/diydrones/ardupilot/pull/2609
Closing.
Thanks, Grant.

closed 2015-07-28 by gmorph