ArduPilot/ardupilot #2596: Traditional Heli Swash setup

Jman841, 2015-07-23:
When setting up the Swash on traditional heli, there is no way to completely lock the swash at center for mechanical setup and trimming servos with the Tx on. The only way to do this is to select manual, then, with trial and error using the Tx stick to reach the PWM value you selected as center.

It would make setup much easier if there was a button that locks the swash at Center on the collective, one at Max pitch and one for minimum pitch.

Jman841, 2015-07-23:
Or being able to lock the swash in manual without it changing the offsets for adjusting max and min PWM limits for the collective. With the current method, if you choose manual and accidentally go past the proper limits, it changes the limits and is very frustrating trying to set up the swash properly.

R-Lefebvre, 2015-10-13:
PR to mostly solve this issue here:

https://github.com/diydrones/ardupilot/pull/2971

It gives a mode where the swash is locked in center, but it does not provide an option for full up/down collective.  However, that can easily be done by leaving it in Mode 1 Manual Passthrough, and putting the stick full up/down.

R-Lefebvre, 2015-12-15:
Closing issue, this feature is in Master and AC3.3 now.

closed 2015-12-15 by R-Lefebvre