ArduPilot/ardupilot #2602: Obstacle Avoidance with GPS waypoints in APM 2.6

siddh2010, 2015-07-24:
Hi guys,
I was wondering if there was a flight mode that used IR sensors for obstacle avoidance along with the regular waypoint navigation. If not, how would I go about adding it and building it into a android app.
Thanks

MaxxUAV, 2015-07-24:
I think this mode do not exists for horizontal avoidance, only vertical range control with IR, Sonar (but for sure no WP nav involved).
One thing "simple" would be to create a No-Fly Zone polygon on APP and implements exclusions on Ardu like the Geofence behavior.

siddh2010, 2015-07-24:
What if I connect an Arduino Uno, along with some IR sensors to the APM and process the obstacle avoidance through the Arduino and then tell APM to move right or left. Then the APM can continue with the GPS waypoint mode like normal. Is this possible? How would the arduino tell the APM to mov e right or left.

MaxxUAV, 2015-07-24:
You have several solutions : the most raw one is to intercept RC PWM from receiver, process corrections with your sensors and output SBUS to APM. But be careful with Timers, I would recommend more powerful chips than Uno (I got stucked trying to implements the same solution three years ago, I had 4 Infras, 1 Sonar and 6 PPM input, 9 PPM outputs)
The second solution is to interact in Mavlink with APM by CMD messages(no hardware needed).
But There may be other solutions I didn't noticed !
Leave me a mp if you need more infos : maxime.lafleur@outlook.fr

siddh2010, 2015-07-24:
Ok thanks. Any tips on setting up a mavlink and how would you send
correction messages.

>
> You have several solutions : the most raw one is to intercept RC PWM from
receiver, process corrections with your sensors and output SBUS to APM. But
be careful with Timers, I would recommend more powerful chips than Uno (I
got stucked trying to implements the same solution three years ago, I had 4
Infras, 1 Sonar and 6 PPM input, 9 PPM outputs)
> The second solution is to interact in Mavlink with APM by CMD messages(no
hardware needed).

rmackay9, 2015-07-25:
This looks like a dev question so I'd recommend bringing it up on the drones-discuss email list.  A few of us (Rob, Leonard, myself) have put some thought into how a good algorithm should work so we could give advice.

I think a direct sensor output -> lean angle conversion will only work in very simple situations (althold while the vehicle is only moving slowly) but it's unlikely to work well enough to be adopted by the wider community.  So if you want to make steps for the high quality solution, please ping us on drones-discuss.

closed 2015-07-25 by rmackay9