ArduPilot/ardupilot #2614: Frsky telemetry in S-Port very slow when armed

adfolfotregosa, 2015-07-28:
Running Frsky telemetry in D-Port mode, copter disarmed vs armed has the same telemetry update speed. I was using this setup since frsky telemetry was introduced with a d4r-ii rx + taranis x9d.
 
I changed rx for a frsky X6R , configured pixhawk to S-Port mode and with copter disarmed it woks fine. After I arm the copter the telemetry slows to a crawl. I double checked and in D-Port it works correctly in my setup.

Video showing what I mean:
https://www.youtube.com/watch?v=PCgQ8cLpEvQ

adfolfotregosa, 2015-07-28:
Opentx 2.0.17, no lua script running just stock telemetry. Pixhawk running 3.3_RC5 , not RC8, So the problem at least on my setup existed before. I just wasn't using my X6R rx at that time.

D-Port with D4R-II RX:
https://www.youtube.com/watch?v=cOiFmxIuWbg

S-Port with X6R:
https://www.youtube.com/watch?v=7KSJynEfcRY

Again I just swaped RX and changed pixhawk from D-Port to S-Port.

magicrub, 2015-07-31:
is this only on copter or does it effect all platforms?

adfolfotregosa, 2015-07-31:
Matthias discovered the problem. It is caused by logging. If the frsky S-Port implementation is the same in all platforms them yes.

https://groups.google.com/forum/m/#!topic/drones-discuss/uG7RZ8t7ZXw

magicrub, 2017-05-18:
is this still valid?

auturgy, 2019-11-18:
@yaapu can you let me know if I can close this?

yaapu, 2019-11-18:
this looks like a duplicate of https://github.com/ArduPilot/ardupilot/issues/6274 which is still open waiting for "independent" confirmation