ArduPilot/ardupilot #2615: Compass variance on rc7 and rc8

andrea25, 2015-07-28:
I have this board:
http://www.goodluckbuy.com/px4-pixhawk-lite-v2-4-5-32bits-open-source-based-pixhawk-compatible-pixhawk.html

I have tried it with AC3.2.1 and all was fine.
I have tried it with AC3.3 rc7 and rc8 and I have problems:
- in the hud it display compass variance. I attach the screenshot.
[image]

Why with rc7 and rc8 I have the problem and with AC3.2.1 I didn't have it?

rmackay9, 2015-07-28:
Best to report this on the AC3.3 beta testing thread and please include a dataflash log if at all possible.  I'll close this because we can capture and deal with these issues on the testing thread.  thanks for the report!

closed 2015-07-28 by rmackay9

andrea25, 2015-07-28:
Ok, thank you!