ArduPilot/ardupilot #2616: Segmentation fault bbbmini

awright424, 2015-07-28:
I have been having issues with ardupilot segfaulting after compiling it. I am compiling directly on my beaglebone black and have tried building arduplane and arducopter and they both segfault when trying to run the binary. I initially got around this a week ago by switching to the copter-3.3 branch but it is now segfaulting as well when running it as well. Everytime I have compiled them using "make bbbmini". I would love to know what is causing this.

magicrub, 2015-07-28:
need to upgrade c++ to version 4.7 or 4.8

awright424, 2015-07-28:
g++ --version
g++ (Debian 4.9.2-10) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

debian@arm:~/ardupilot/ArduCopter$ gcc
gcc             gcc-ar          gcc-nm          gcc-ranlib
gcc-4.9         gcc-ar-4.9      gcc-nm-4.9      gcc-ranlib-4.9
debian@arm:~/ardupilot/ArduCopter$ gcc -version
gcc: error: unrecognized command line option ‘-version’
gcc: fatal error: no input files
compilation terminated.
debian@arm:~/ardupilot/ArduCopter$ gcc --version
gcc (Debian 4.9.2-10) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

I dont think thats the issue. Do I need to downgrade?

magicrub, 2015-07-28:
oh, well then this is something new. I don't have a BBBmini so I can't
check - sorry!

awright424, 2015-07-28:
Running the ArduCopter.elf binary with sudo was not providing any output so I only stumbled across the segmentation fault by running directly as root.

awright424, 2015-07-28:
no I have stumbled across more by running with this with gdb.

0x0002ca22 in Linux::LinuxI2CDriver::LinuxI2CDriver (
    this=0xa9520 <i2cDriver0>, semaphore=<optimized out>, 
    device=<optimized out>)
    at /home/debian/ardupilot/libraries/AP_HAL_Linux/I2CDriver.cpp:36
36	        hal.scheduler->panic("I2C device is not a chardev node");

magicrub, 2015-07-31:
does anyone have any insight on this?

earsneyes, 2015-07-31:
@magicrub there is a discussion about here https://github.com/mirkix/BBBMINI/issues/10

juvinski, 2015-08-01:
This was caused by wrong i2c device pointed by the code(i2c-2) instead(i2c-0 or 1).
Check what i2c devices you have in your board(ls /dev/i2c-*) and them change on code.

awright424, 2015-08-03:
This has been fixed. It was an issue with using the wrong i2c device.

closed 2015-08-03 by awright424