ArduPilot/ardupilot #2622: NMEA GNSS receiver initialisation message

suraqis, 2015-07-29:
When using other NMEA GNSS receiver than specified in code (SIRF, MTK, U-Blox)  autopilot tries to configure it with initialisation messages. Some receivers can't ignore that messages and fail their initialization even already configured. We need to have "generic NMEA" in order to not send any initialization messages when initializing GNSS receiveer.

rmackay9, 2015-07-29:
I think you should be able to set the GPS_TYPE parameter to specify that you only want to use NMEA GPS and then it will only send the init string for the NMEA.  I could be wrong but I'm wondering if you've tried this to see if it works?

suraqis, 2015-07-29:
I set GPS_TYPE to 5 (NMEA) and it sends something to my receiver that makes it crazy. Now I just disconnect tx pin of autopilot's serial port to have it working.

WickedShell, 2015-07-30:
What receiver? The entire auto detection/configuration sequence is dependent upon the reciever ignoring any message it doesn't understand. All the configure stuff sent at the start is a binary message for ublox and then a bunch of P messages for ublox, sirf, mtk etc.

WickedShell, 2015-10-29:
While you're GPS should ignore the other messages coming in as they are all manufacturer specific messages/binary you're also correct that the entire auto config blob is sent regardless of what GPS type you select. I renamed the issue to make it clearer.

suraqis, 2015-10-29:
Sorry for ignoring previous question. That receiver was Novatel OEM module based device. It sends correct NMEA sequences but can't ignore config strings. I'd like to have "NMEA without configuration" type of GNSS receiver. That means that user should correctly setup receiver for NMEA operations.

WickedShell, 2015-10-29:
Yeah there isn't a NMEA config string, at least not on the autodetect level which is what I think was messing you up. All the config strings are for the other drivers... I don't actually know whats in the NMEA driver after that, but lets start with the first step of not sending the autodetect.

mr337, 2016-08-08:
I agree, this just bit me yesterday and is messing up my Eclipse receivers. I'll be whipping up a PR to not send the config string. Initially I thought the GPS_AUTO_CONFIG was an issue but looking at the code is only used for UBLOX recievers. Everything gets send the blob regardless.

commit 51e8d8a2 (2016-08-16): AP_GPS: Don't send config blob to GPS unless AUTO_CONFIG=1

All GPS types will recieve startup up blob config. For
some recievers this will cause them to hang. This commit
only allows sending of blobs if AUTO_CONFIG=1. Fixes #2622
closed 2016-08-16 by tridge