ArduPilot/ardupilot #2623: Bad Lidar health Message when out of range

Jman841, 2015-07-29:
Currently when flying with the LIDAR Lite on Arduplane 3.3.0 when you go past the range of the LIDAR (40 meters) the GCS reports "Bad Lidar Health" for the entire duration of the flight until you are back under the 40 meter mark. This is quite annoying during missions as it persists the entire mission.

Also, When using the voltage stop pin to turn off the LIDAR Lite above a specified height. There is currently no way that I am aware of to verify that the LIDAR Lite is actually powered off as the GCS reports the last voltage received from below 40 meters. If the Lidar voltage could go to 0 and the autopilot recognize that it is not Bad LIDAR health and is normal function of being above it's range it would make things much more comfortable during the missions.

Thank you

magicrub, 2015-07-31:
the lidar lite is rated for 40m, but not all units are equal and it certainly depends on how reflective the surface is that it's looking at. I use a range limit of 1.5m to 30m in PWM mode and don't get those alerts doing flight. Please try that and report back. 

On a separate matter, on the ground I get the same message you get and it's quite annoying.

Jman841, 2015-07-31:
Ok, I will give it a try.

Also, is there any way to confirm that the power has been cut to the Lidar unit over a specific height?

magicrub, 2015-07-31:
no, I ended up putting an LED (or o-scope) on the enable pin

jinchengde, 2015-08-04:
maybe you could try to set RNGFND_LANDING to 1

Jman841, 2015-08-04:
I have RNGFND_LANDING set to 1. with it set to 0 it will always read bad lidar health. I only get bad lidar health messages when the sensor is out of range. 

I am going to test with the max range value set below 40 meters as Tom suggested and set the stop pin for 40 meters and see if that fixes the issue. I'll report back after testing.

closed 2021-07-26 by Naterater