ArduPilot/ardupilot #2638: Traditional Heli Auto-Rotation failsafe

Jman841, 2015-08-01:
Would it be possible to have a failsafe for traditional heli's utilizing an auto-rotation function? 

Thinking a little bit about it, it would seem to need 2 sensors. First, and most importantly, a LIDAR Lite or precise range finder sensor as any baro drift would probably be too inaccurate for a solid auto-rotation. and Second, an RPM sensor for the motor to tell if the motor has stopped spinning and this failsafe is required. Many ESC's offer this function natively and if there is development for a pixhawk governor it may already be plugged into the system.

JeffreyMeyer, 2017-05-23:
Jman841 commented on Aug 1, 2015:
"... if there is development for a pixhawk governor it may already be plugged into the system."

Can anyone tell me if pixhawk does indeed handle autorotation?

rmackay9, 2017-05-23:
ArduPilot doesn't support auto-rotation although we've talked about it a few times.

auturgy, 2021-02-10:
@Gone4Dirt @bnsgeyer can this be closed?

bnsgeyer, 2021-02-10:
@auturgy  No.  We have started development and @Gone4Dirt has incorporated the beginnings of this into SITL.