ArduPilot/ardupilot #2651: Capability to handle LIDAR LITE 2 I2C

roque-canales, 2015-08-03:
Hello,
 
I bought one lidar lite 2 and I tried to connect it to AC3.3 RC8 via i2c following wiki, and it no work.

Could you please add I2C driver for this new lidar please?

magicrub, 2015-08-03:
Rumor has it that @zlite has one and someone at 3DR is already working on it. I'm not sure of the merge and/or development plan but it would be great it the v2 driver update would make it to diydrones  master and not get stuck in a solo branch.

silverburn, 2015-08-04:
Put me down for one of these too...bought the V2 thinking it fixed all the bugs in V1, but only to find it needs a code update, and it's another expensive paperweight, just like the V1.

Jman841, 2015-08-04:
The V1 works quite well in PWM mode. Shouldn't be a paperweight

silverburn, 2015-08-04:
I don’t have any PWM slots available though…needs to be I2C. :-(

magicrub, 2015-08-04:
Works great in pwm mode. So far in 3 units no GPS jamming at all.

rmackay9, 2015-08-05:
@silverburn, you know the AUX ports can be repurposed as PWM inputs?  It's one of the BRD_ parameters

magicrub, 2015-08-05:
Here's the guide
<http://plane.ardupilot.com/wiki/common-optional-hardware/common-rangefinder-landingpage/common-rangefinder-lidarlite/>

silverburn, 2015-08-05:
Guys, yes, I've been through all of this, but I have no free aux slots left in the Pixhawk, which is why I needed I2C to work.

lucasdemarchi, 2015-08-05:
Just a note that it works great in Linux using the kernel driver. It will probably be available in v4.3 or v4.4: http://www.spinics.net/lists/linux-iio/msg20126.html

Ideally we could use the kernel driver rather than doing it in userspace. So if anyone is willing to write the glue code would be greatly appreciated for Linux boards. @staroselskii I see there's some work in progress about LIDAR in Linux [here](http://community.emlid.com/t/lidarlite/457/11). Would you guys be interested in switching to use the kernel driver instead?

staroselskii, 2015-08-07:
@lucasdemarchi 

It's definitely worth a try. At the moment I can't test it myself, though. I remember Tridge had pointed out some issues with using kernel drivers. I believe, it has something to do with sensors' synchronization. 
Correct me if I'm wrong, the driver at hand implies the use of blocking reads, doesn't it? That's why there should be any significant synchronization issues, right?

lucasdemarchi, 2015-08-07:
@staroselskii I'm not aware of any issues. You still must know what you are doing (just like doing it in userspace): if you have multiple sensors in the same bus you will likely have issues with not having enough bus bandwidth... and you can't mix userspace and kernel space drivers on the same bus.

silverburn, 2015-08-07:
Wow, I think I’m a bit out my depth here…

I’m still debugging my basic script: 10 print “hello world”; 20 GOTO 10….:-)

Still, anything I can do to help get a working driver for pixhawk, just give us a shout!

Euan

rmackay9, 2015-08-08:
Jaime Machuca's going to try to develop the driver so hopefully he'll pop up when he has something for us to test.

rmackay9, 2015-08-17:
Support for the lidar-lite V2 via I2C is in master now so closing.  Apparently Tridge has found some issues with the sensor suddenly reporting a distance which is off by about 13m but so far we think that's a sensor issue not an ardupilot driver issue.

Hopefully this will go out with AC3.3-rc9 in case you're a copter user.  As long as the merge conflicts aren't too terrible that's what we will do.

closed 2015-08-17 by rmackay9