ArduPilot/ardupilot #2655: AC3.3 Dataflash log issue

roque-canales, 2015-08-04:
Hello,

I experienced the logs on RC7 and RC8 and I was unable to disable though log_bitmask parameter  theses very heavy logs (20Mo for 10min) :VIBE,UBX1,UBX2,UBX3,IMT,IMT2,IMU,IMU2

Could you please give us new log_bitmask matrix / registered logs?

rmackay9, 2015-08-04:
Are you sure this applies to -rc8?
In -rc7 we forced the higher logging rate required for detailed analysis of the EKF but that fix was not included in -rc8 I believe so it should be possible to reduce the logging rate.  I'm going to close this issue but we can re-open if you double check -rc8 and indeed I'm wrong.

WickedShell, 2015-08-04:
I know at least in the GPS code UBXn will always come out no matter what logging is set to. It drives me insane. I was going to add a parameter for awhile, but with the proposed DF changes I think it becomes possible to be a bit more sane about whats logged there. But this is an all vehicles thing on at least the UBX stuff.

Repoening for the GPS subpart at least.

rmackay9, 2015-08-04:
@WickedShell, ok, but I think this is a different issue that you're talking about.  The UBX messages can't be contributing that significantly to the log file size in comparison to the 400hz IMU logging that was forced on by default in -rc7.

I've created a new issue for the UBXn messages and closing this one.

closed 2015-08-04 by rmackay9

roque-canales, 2015-08-07:
Randy I confirm that all these high rate logs (VIBE,UBX1,UBX2,UBX3,IMT,IMT2,IMU,IMU2) are continuously recorded on COPTER 3.3 RC8 even if we go to console put LOG DISABLE command.

So, could you update the log bitmask to be able to disable all these logs?