ArduPilot/ardupilot #2656: Allow disabling UBX dataflash messages

rmackay9, 2015-08-04:


roque-canales, 2015-08-07:
Randy I confirm that all these high rate logs (VIBE,UBX1,UBX2,UBX3,IMT,IMT2,IMU,IMU2) are continuously recorded on COPTER 3.3 RC8 even if we go to console put LOG DISABLE command.

So, could you update the log bitmask to be able to disable all these logs?

WickedShell, 2017-06-09:
How do people want to be able to disable these messages?

I'm not excited at yet another GPS lib parameter, but that is the precedent with the GPS_SBP_LOGBITMASK. We could rename that to be GPS_LOGBITMASK and overload it to have meanings for UBlox, but it wasn't originally written in a way that supports that use case nicely.

OXINARF, 2017-06-11:
I'm more concerned about messages being written even when disarmed - but I'm not sure if that is still the case with master.

Regarding being able to disable logging them, is a bitmask all that useful? @njoubert? I'm thinking that a simple parameter saying if driver specific messages should be logged or not would be enough.

WickedShell, 2017-06-11:
I don't need a bitmask for the u-blox stuff, I'm just aware thats what SBP is using atm. I'm happy to go either way.

Regarding logging while disarmed it checks that  ```gps._DataFlash->logging_started()``` is true, so I think it means we don't log while disarmed anymore if we aren't supposed to....

peterbarker, 2017-12-19:
UBX messages don't appear now.

closed 2017-12-19 by peterbarker