ArduPilot/ardupilot #2664: Copter: Change mount_mode with DO_MOUNT_CONTROL

squilter, 2015-08-06:
DO_MOUNT_CONTROL should consume parameter 7 and change the mount_mode.  That way, arducopter can do a DO_MOUNT_CONTROL in guided mode, not just auto.

squilter, 2015-08-10:
@rmackay9 ArduCopter is using `MOUNT_CONFIGURE` and `MOUNT_CONTROL`.  These are similar, but not the same as `MAV_CMD_DO_MOUNT_CONFIGURE` and `MAV_CMD_DO_MOUNT_CONTROL`.  The 7th parameter that we wanted to consume is only in `MAV_CMD_DO_MOUNT_CONTROL`, which isn't a packet we are listening to.

rmackay9, 2015-08-10:
Ok, I'd suggest that we should get rid of the ardupilotmega messages and stick with the common messages.  It's a pain that that will cause changes to all ground stations and vehicles though!

squilter, 2015-08-10:
@rmackay9 So these are actually totally different messages?

rmackay9, 2015-08-10:
@squilter yes, I'm afraid so.  One is sent within a COMMAND_LONG or MISSION_ITEM, the other two are their own individual messages.

squilter, 2015-09-09:
Done with #2696

closed 2015-09-09 by squilter