ArduPilot/ardupilot #2665: ArduCopter-3.2.1 branch does not compile succesfully

luo674526417, 2015-08-06:
%% MODULE_MK           = /home/lc/px43.2.1/PX4Firmware/src/modules/px4iofirmware/module.mk
%  MODULE_NAME         = px4iofirmware
%  MODULE_SRC          = /home/lc/px43.2.1/PX4Firmware/src/modules/px4iofirmware/
%  MODULE_OBJ          = /home/lc/px43.2.1/PX4Firmware/Build/px4io-v2_default.build//home/lc/px43.2.1/PX4Firmware/src/modules/px4iofirmware/module.pre.o
%  MODULE_WORK_DIR     = /home/lc/px43.2.1/PX4Firmware/Build/px4io-v2_default.build//home/lc/px43.2.1/PX4Firmware/src/modules/px4iofirmware
make[3]: Entering directory `/home/lc/px43.2.1/PX4Firmware/Build/px4io-v2_default.build/home/lc/px43.2.1/PX4Firmware/src/modules/px4iofirmware'
make[3]: Nothing to be done for `module'.
make[3]: Leaving directory `/home/lc/px43.2.1/PX4Firmware/Build/px4io-v2_default.build/home/lc/px43.2.1/PX4Firmware/src/modules/px4iofirmware'
LINK:    /home/lc/px43.2.1/PX4Firmware/Build/px4io-v2_default.build/firmware.elf
BIN:     /home/lc/px43.2.1/PX4Firmware/Build/px4io-v2_default.build/firmware.bin
%% Generating /home/lc/px43.2.1/PX4Firmware/Build/px4io-v2_default.build/firmware.px4
fatal: Not a git repository: '/home/lc/px43.2.1/PX4Firmware//.git'
make[2]: Leaving directory `/home/lc/px43.2.1/PX4Firmware/Build/px4io-v2_default.build'
%% Copying /home/lc/px43.2.1/PX4Firmware/Images/px4io-v2_default.px4
%%%%
%%%% Building px4fmu-v2_APM in /home/lc/px43.2.1/PX4Firmware/Build/px4fmu-v2_APM.build/
%%%%
%  PX4_BASE            = /home/lc/px43.2.1/PX4Firmware/
%  GIT_DESC            = "unknown_git_version"
%  CONFIG              = px4fmu-v2_APM
%  BOARD               = px4fmu-v2
%  WORK_DIR            = /home/lc/px43.2.1/PX4Firmware/Build/px4fmu-v2_APM.build/
%  NUTTX_EXPORT_DIR    = /home/lc/px43.2.1/PX4Firmware/Build/px4fmu-v2_APM.build/nuttx-export/
%  NUTTX_CONFIG_HEADER = /home/lc/px43.2.1/PX4Firmware/Build/px4fmu-v2_APM.build/nuttx-export/include/nuttx/config.h
make[2]: Entering directory `/home/lc/px43.2.1/PX4Firmware/Build/px4fmu-v2_APM.build'
/home/lc/px43.2.1/PX4Firmware/makefiles/firmware.mk:221: *** Can't find module(s): drivers/batt_smbus drivers/ll40ls drivers/px4flow.  Stop.
make[2]: Leaving directory `/home/lc/px43.2.1/PX4Firmware/Build/px4fmu-v2_APM.build'
make[1]: *** [/home/lc/px43.2.1/PX4Firmware/Build/px4fmu-v2_APM.build/firmware.px4] Error 2
make[1]: Leaving directory `/home/lc/px43.2.1/ardupilot'
make: *** [px4-v2] 错误 2

This is what causes, please explain!!!

rmackay9, 2015-08-06:
This is a dev question so probalby best to send to drones-discuss@googlegroups.com or ask on the ardupilot gitter.

closed 2015-08-06 by rmackay9

luo674526417, 2015-08-06:
but ArduCopter-3.15can branch succesfully!!!

luo674526417, 2015-08-06:
@rmackay9 but ArduCopter-3.15can branch succesfully!!

luo674526417, 2015-08-06:
@rmackay9 I use ubuntu 12.04  branch  thanks!