ArduPilot/ardupilot #2668: Function documentation

lucasdemarchi, 2015-08-06:
The codebase is very inconsistent with regard to where we document the functions: header or implementation. For new code it'd be good to always have the comments documenting a method in either of them and update our [Coding style guide](http://dev.ardupilot.com/wiki/style-guide/).

I don't have a strong opinion on either of them... I've already seen several projects using one or another.  The bad thing about the header is that it "pollutes" a lot, but it's the most visible place.

thoughts?

magicrub, 2015-08-06:
Personally, I prefer the implementation or at least the space just above
the function implementation. Also, standardizing a few things such as
describing each input variable and it's units and describing what the
return is, if any.

rmackay9, 2015-08-07:
I tend to copy the comments into both places.

WickedShell, 2015-08-07:
I tend to like both. Function for when in the code, header for checking the
expected result and interface. I also admit that I'm a fan of doxygen (or
similar) on a well documented project, especially when getting started on
it.

It's a massive effort to get it right but I'd love to see functions self
document if they side effect. That has really messed code up for me in the
past, and is (or at least feels) so inconsistent in the project it would be
nice to document.

rmackay9, 2015-08-07:
We have a really really old PR in which someone tried to start adding doxygen comments to the code.
https://github.com/diydrones/ardupilot/pull/498

We didn't merge it back then, it's super out of date so I'd like to close it but I think we can only do that if we have an alternative strategy including at least a small amount of work by someone to sort out some of the comments and a plan to continue through the rest of the code at some point.

magicrub, 2015-08-07:
+1 for doxygen, great suggestion Michael.

lucasdemarchi, 2015-08-07:
> We didn't merge it back then, it's super out of date so I'd like to close it but I think we can only do that if we have an alternative strategy including at least a small amount of work by someone to sort out some of the comments and a plan to continue through the rest of the code at some point.

I think we must first agree on one of them and enforce it for **new** code. If we wait on another big pr adding the comments we will never actually reach that point because it's such a huge effort that gets very easy outdated.

So my suggestion/vote from what I've read on this issue: add doxygen comments to the function implementation.

WickedShell, 2015-09-09:
In my not so copious spare time I've started to look down the doxygen road. I don't feel strongly invested in doxygen as the required tool, but I'd like to take a look at starting to include a tool of this type. I think it would add a lot of value, and it definetly would have helped with some of the changes I've made in the past.

Is everyone good with starting to include doxygen comments on all new functions? (And somewhat slowly add comments over old code). Doxygen etc only works if it becomes a requirement for all new commits, it will raise the effort on a per commit basis a bit, but adds a lot of value to the code base, at least in my opinion.

rmackay9, 2015-09-10:
@WickedShell, it's worth a try.  Maybe just start with one file and then we can discuss on the dev call?

lucasdemarchi, 2015-09-11:
On my last commits I tried doxygen and remembered how verbose it is :(. However it's the most common and one of the easiest to setup.

However in order to be successful I think we should agree on:
 - Agree on where the documentation goes; doing both on cpp and header would be the wrong thing IMO since it does get out of sync. Doing only on cpp has the benefit of a) not cluttering the header and b) being near to the implementation, but has also a problem: what about inlined methods?
 - Agree to use only one of the syntax doxygen offers
 - Agree on what to do for inherited methods. Most of the methods implementing a pure virtual method would be fine just inheriting the doc. But then again, the documentation would need to be in the parent class, in the header (since it's pure virtual)
 - Enforce for new commits

I think we can talk about this on the dev call in order to decide it. Having some examples at hand would be nice, too.

hamishwillee, 2015-09-22:
+1 

1. Typically behaviour is re documented/copied in inherited methods. A bit of a pain.
1. Most projects I have worked on prefer API doc in the headers rather than cpp. It is really a matter of preference though.
1. Good to document for functions - parameters, return types, exception thrown, default values, preconditions/assertions. 
1. No idea about how it is for ArduPilot, but in most projects there has been a lot of emphasis on documenting and defining a "public API". Ie the bits that we promise to document, keep up to date, and try not to change.
1. I wouldn't necessarily say that the whole system needed to be documented in one go, but you should ensure that complete classes/files are documented in one go. Hard to track otherwise.

lucasdemarchi, 2015-09-25:
Now that we had enough discussion about it I think we can close this issue.

closed 2015-09-25 by lucasdemarchi

hamishwillee, 2015-09-28:
@lucasdemarchi Just FMI, and for anyone else coming "late to the party" what was the result of that discussion?