ArduPilot/ardupilot #2669: Plane: Stall Recovery

Jman841, 2015-08-06:
One thing Ardupilot is terrible at in any auto or FBW mode is recovering from a stall if it does occur. While the stall prevention code works excellent and the chance of a stall is minimal. I have stalled a plane in FWBA and heard of stalls in auto mode. The issue is, if you do not switch to manual and do a simple stall recovery (gain speed then elevator up) currently arduplane will make the model death spiral with no chance of recovery.

Could we add a safety feature where a stall is recognized then, if airspeed sensor is present, it waits for airspeed to get above the minimum and then simply applies elevator to recover?

I realize all planes stall differently and some may not be able to be recover, but, the current response from Arduplane is a guaranteed crash.

magicrub, 2015-08-06:
+1

As a flying wing pilot, my stalls are harder than usual to recover from. If
ArduPilot could recover from that then we can probably recover from any
frame type.

I've had many stalls during MANUAL which are human error, but I've had
several stalls during AUTO. In one case it was a poorly calibrated (zeroed)
airspeed sensor which I woudl call human-error but two other ones where It
launched with good calibration and during a flight, or rather after it
crashed, the airspeed was stable at 7 or 8 meaning the calibration changed
during flight which caused the stall.

Anyway, yes, I'm totally in favor of us creating a stall DETECTION and
RECOVERY feature.

jinchengde, 2015-08-07:
We also meet the crash in Geo-fence, if the plane dive and lower than the minimum altitude , the geo-fence in work and take the plane to the return point, but sometimes, the plane can't recovery from dive attitude immediately , then crash directly.
So maybe we could use DETECTION to find the dangerous situation and RECOVERY
1. level the plane
2. power maximum
3. pull up the plane

gmorph, 2015-08-07:
I'd be interested in seeing the dataflash log (downloaded from your autopilot) from where you hit the geofence and the plane didn't recover.
Thanks, Grant.

Jman841, 2015-08-07:
After thinking about this for a bit, I think its application should be independent of the airspeed sensor as many of the stalls are due to an error/fault with the airspeed sensor.

geofrancis, 2015-08-19:
the problem I found with a stall in auto modes is that the autopilot keeps pulling up to try and bring it level but it cannot due to the lack of airspeed so it just keeps falling.

first it would have to know it was stalled like if its loosing altitude with the nose down and max throttle and elevator for more than a couple of seconds then all it would have to do is level the elevator for a second before trying to pull up again.

magicrub, 2015-08-19:
"the autopilot keeps pulling up to try and bring it level"
@geofrancis That's not what ArduPiIot does, are you flying with a pitot tube? Do you have any logs showing this behavior? During a stall if it keeps falling that's because you're already stalled and the flight surfaces aren't working in the usual sense and there's currently no stall recovery logic.

Stall prevention (supported):
If airspeed is lower than expected, throttle up and pitch down.

Stall detection (not implemented):
detect flat spin? ultra low air-speed? unexpected high sink rate?

Stall Recovery (depends on Stall detection):
if in a stall, do something to get out of it which is plane-type dependent. Normal aircraft with elevator and rudder would do something like apply rudder to stop a spin (if you're spinning). In a flying wing it may get complicated. What about copter? What about no ailerons (rudder only)?

geofrancis, 2015-08-19:
what i mean by "the autopilot keeps pulling up to try and bring it level" is that when the plane is in a tip stall spiralling nose down the auto pilot does not know its in stall so its always trying to pull the nose up, this causes the plane to keep spiralling down all the way into the ground since it will never gain enough airspeed to regain control again.

R5fan, 2015-09-28:
geofrancis is spot on. Ive also encountered a spin while doing autotuning and its only because I switched to manual quickly enough that I didnt lose the plane. That plane has pretty benign stall characteristics, its only because the fc actively tried to counter the spin, so tried to pull up the nose and level the wings, exactly what should never do in a spin,  that it remained in a spin in the first place. 

Im not sure how to best recognize a spin though, every plane is different. Some will spin with the noise pointed nearly vertical, others can do much flatter spins. In either case though, you should be abe to infer it from the magnetometer, especially if you combine it with decreasing altitude and non responsiveness from roll and pitch controls.

magicrub, 2015-10-02:
This depends on https://github.com/diydrones/ardupilot/issues/2933

geofrancis, 2016-03-25:
i have been doing some testing with my little Z84 wing with its APM. I found that its nearly impossible to get into an uncontrolled stall while its in acro mode since it never tries to pull out the dive it just stops the rotation. so after some testing i found when the plane is in an uncontrolled stall in stabilised mode I flick the switch to acro for around a second, this immediately stops the spinning then when i flick back to stabilised it pulls the nose up back to level. 

for most aircraft to recover i think one second of acro mode would be enough to recover most aircraft before enaging a mode with autolevel to bring it back to flat and level.  

if the stall detect logic could be made to work then it would be a good way of recovering.

marcmerlin, 2016-04-05:
@geofrancis you said in your other issue that stablize, you could not get out of a flat spin because the AP did not lower the nose to regain airspeed and  aileron control authority.
Did you have an airspeed sensor? AP ought to know to pitch down to regain airspeed and get out of a stall with an airspeed sensor.

geofrancis, 2016-04-05:
i have been flying wings with apm recently and have been testing stall recovery and how to automate it.

when i would switch to stabilised and hold the nose up with no throttle it would go into a flat spin and spiral nose down until it crashed no matter how much throttle you gave it since it would keep trying to pull up.

So after experimentation and lots of crashes i found the most reliable technique is to switch to acro mode for 2 seconds with the throttle at 100% then to RTL. this would stop it trying to pull the nose up long enough for it to stop the spin and pick up airspeed so when it switches to RTL it can pull the nose up and back to stable flight.

I programmed this in opentx so when i hit the spring loaded trainer switch it recovered but it shouldnt be to hard to implement as a flight mode in arduplane so when you get stall detection working.

also if you assign an input for a stall sensor and just have a parameter for it so if its above a specific value then plane is stalled then it would let people design sensors that can be simply plugged in.

WickedShell, 2016-04-05:
I haven't tested this procedure yet, but it should be noted that power on
is generally an awful thing to do to a spin recovery so that would need to
be differentiated from a stall. What airframe were you using? (IE flying
wing or a conventional plane?)

geofrancis, 2016-04-05:
it was with a  Z84 flying wing. i tried recovering several times without any power and it would not respond until i gave it full power. next time i go out i will save the flight logs

marcmerlin, 2016-04-05:
@WickedShell 
The FAA pilot handbook (and my instructors) disagree with you (cutting power for stall recovery)
https://www.faa.gov/regulations_policies/handbooks_manuals/aircraft/airplane_handbook/media/faa-h-8083-3a-3of7.pdf page 4-4
"Second, the maximum allowable power should be applied to increase the airplane’s airspeed and assist in reducing the wing’s angle of attack. The throttle should be promptly, but smoothly, advanced to the maximum allowable power. "
Spin recovery is only different from stall recovery in that you add rudder to stop the spin (assuming you have a conventional airplane that recovers from spins, some planes/airframes do not)

WickedShell, 2016-04-05:
@marcmerlin I specifically meant for spin, not stall.

Quoting from you're source for spin recovery.
"Step 1—REDUCE THE POWER (THROTTLE) TO IDLE. Power aggravates the spin characteristics. It usually results in a flatter spin attitude and increased rotation rates."

WickedShell, 2016-04-05:
Page 4-15 of that PDF.

geofrancis, 2016-04-05:
thats only relevant to standard aircraft, wings dont have rudders and arduplane does not stabilise with the rudder at all.

WickedShell, 2016-04-05:
Yup, that's my point actually is that we need to differentiate the recovery
techniques for the type of airframe :)

geofrancis, 2016-04-05:
well my technique works with wings, but i think it would work with most stalling aircraft the main problem currently is that it always wants to pull up, on my bixler 2 all i had to do was switch to manual briefly and it would immediately recover when switched back. since all it needed to do was stop pulling up to regain control.

WickedShell, 2016-04-05:
Right, I'm just trying to observe that it is generally a bad recovery
technique for non wings that are in a spin.

geofrancis, 2016-04-05:
your missing the point that the only reason it stays in the spin is because the flight controller is has the elevator at maximum and wont put it down so it can recover. thats where acro comes in since it wont try and pull up, letting the plane recover.

WickedShell, 2016-04-05:
That's not the only reason a standard aircaft would stay in a spin, the
applied throttle while also aggravate the spin and can cause it to
increase, thus that needs to be accounted for as well on a non wing, that's
all I'm trying to observe here.

geofrancis, 2016-04-05:
the current problem i am trying to address is the flight controller holding max elevator while stalled. it effects all aircraft, i have tested this on a bixler 2 glider and a z84 wing @R5fan also had the same issue, acro mode and manual are the only flight modes that wont fight the stall, both z84 and bixer could be recovered manually but the wing has a habit of snapping into a spin in the opposite direction thats why i now use ACRO for stall recovery since it will stop the rotation automatically without any inputs.

marcmerlin, 2016-04-05:
@WickedShell sorry, I did indeed make the mistake to treat spin = stop spin rotation + stop stall.
In the airplanes I fly, opposite rudder stops the spin so quickly that you don't have time to cut power (nor do you need to).

jcooper21, 2016-04-06:
I many times I have stalled a flying wing and sent it into a death spiral on purpose. I switch to RTL and it pulls out of the death spiral on its own.

geofrancis, 2016-04-06:
what wing are you using?

R5fan, 2016-04-06:
@jcooper21 
While it may seem to work on your particular plane, its a *terrible* idea. It will only recover from a spin if the plane is really benign and conditions right. What ardupilot currently does in a spin is about the worst thing you can possibly do as a pilot (roll out and pull up). Again, if your plane is well behaved and will recover on its own (sticks neutral) or even with incorrect input (ardupulot), so even doing the worst may not crash your plane that day, but its hardly a strategy you want to actively pursue. 

Flip it to manual (or acro) until the spin is recovered from, then use whatever flight mode you want, but dont rely on RTH or any autonomous mode to save you from a spin as on many planes, it wont, and one day, it may not save yours either because your cg is a bit off or whatever.

Jman841, 2016-04-14:
Any update on this? Or on the Stall detection piece? I had a very strange crash today due to a stall caused by a high bank angle and low airspeed in a turn in Auto mode. The plane was commanded to take off to 100 meters, then switch to RTL which is at 150 meters. Once it reached 100 meters it went into a 45 degree climb (Although the limitation is set to 25 degree's) then leveled off into a sharp turn to start the RTL and at this point the airspeed was low, throttle dropped to 60% and the plane tip stalled. Once the stall started, it was the familiar spiral of death that always happens when arduplane stalls. Unfortunately I had turned the RC Controller off shortly before this and it could not turn on in time to switch to manual and save the plane.

magicrub, 2016-04-14:
if you did have manual control available, what would you have done?

marcmerlin, 2016-04-14:
not my plane, but I would have neutralized the elevator, turned the ailerons opposite of stall, which may not work in a spin, but that mixes in rudder too, which does work, and then regain altitude once the spin has stopped.
Although to be honest, all my RC planes stop spinning once I stop pulling the elevator to go up (which likely is what AP was trying to do).

Jman841, 2016-04-14:
neutralized the servos and give it full throttle then apply elevator to pull up. It stalled at 150 meters so if the radio would of turned on faster or if I didnt turn off the radio in the first place I would have had enough time to recover. I believe.

Jman841, 2016-04-15:
Here's the Logs:

https://www.dropbox.com/s/tzsmfcwcf3zvdy9/7%204-14-2016%2010-31-28%20AM.bin?dl=0
https://www.dropbox.com/s/w8seipoxwee3eel/2016-04-14%2010-13-56.tlog?dl=0

iskess, 2016-04-15:
@Jman841, I'm curious to know why you switched off your radio just after takeoff. I'm not being judgmental, I'm just wanting to know why one might do that.

R5fan, 2016-04-15:
@magicrub 
For full scale airplanes, spin recovery procedures can vary, but they all share one thing: forward elevator. Usually this is combined with rudder opposing the spin direction and followed up by adding throttle. As marc merlin pointed out however, RC planes tend to recover easily from a spin, most (especially FPV models) will recover just with neutral sticks, some more aerobatic models may require a little forward elevator to get out (or get out faster). Truthfully, any recovery procedure will likely work fine on the vast majority of rc models, particularly FPV models.

DavidDavidov, 2016-04-15:
Just to add from my experience with the flying wings, FX-61 in
particular. We found that when they enter in flat spin are very
reluctant to recover. Until you put them upside down pushing forward the
elevator stick. That immediately recovers the plane.

dave

On 15-04-16 09:39, R5fan wrote:
> @magicrub <https://github.com/magicrub>
> For full scale airplanes, spin recovery procedures can vary, but they
> all share one thing: forward elevator. Usually this is combined with
> rudder opposing the [...]

WickedShell, 2016-04-15:
I had a very long hash out of this with someone on a different issue but
for spins on full scale airplanes adding throttle in a spin can be really
bad until after the spin has exited, as it has a tendency to tighten the
spin, which is of course bad. Except for some very specific aircraft where
it's called out specifically the general advice for spins is to not add
throttle.

Jman841, 2016-04-15:
@iskess, After the aircraft reaches a safe altitude with this plane we always turn off the radio to avoid mistakes of the user hitting the flight mode switch or any other inputs from the Tx. Plus, most of the missions are beyond the range of the Tx and there is no use for it other than a stall recovery that happens within LOS. This was the first time we've had a stall in an Automatic mode after literally thousands of flights with this airframe. I have a feeling the route cause was a combination of things, primarily that we did not do the airspeed calibration for this area yet (Much lower elevation than we typically flight at), along with this setup being on the heavier side of the models limitations, and finally, I am not sure why the aircraft was allowed to climb at 45 degree's when the params have it restricted to 25. Any idea's why the plane in automatic mode was able to climb at a 45 degree angle?

magicrub, 2016-04-15:
@Jman841 please post a log to the forums and someone will take a look