ArduPilot/ardupilot #2670: Add support for MAV_CMD_DO_MOUNT_CONTROL in guided mode

hamishwillee, 2015-08-06:
Currently, the only way to have full manual camera pointing control, such as tilt on a joystick, is to use an RC Override. It would be better to support MAV_CMD_DO_MOUNT_CONTROL (in all platforms)

Use-case is something like, flying around in Loiter mode, and wishing to look around manually rather than giving aiming points or ROI.

rmackay9, 2015-08-07:
So for copter we support this as a mission command but not an immediate command.  The check for the do-mount-control would need to be added in the GCS_MAVLink.cpp file in the ArduCopter directory.

squilter, 2015-08-07:
@rmackay9 I think this is the same problem that we just talked about, #2664

hamishwillee, 2015-08-07:
@rmackay9 Yes. The implementation should be pretty much the same across platforms - just different GCS_MAVLink.cpp files.

rmackay9, 2015-08-07:
@squilter, yes, i think you're right.  I was searching through GCS_MAVLink.cpp for "DO_MOUNT_CONTROL" but I missed it.  It's actually already there as "MSG_ID_MOUNT_CONTROL".

So for Copter at least, not much needs to be done and what needs to be done is probalby all in issue #2664

squilter, 2015-10-03:
Rover 958d279b7cf69413a34c711028a3b8c96803bfd2
Plane 8c1212d4a6f79f329e1a329baf0e5521e1d6282c

closed 2015-10-03 by squilter

hamishwillee, 2015-10-05:
@squilter @rmackay9 So can you confirm that the change from this to the **documentation** is 

1. enabling the param 7 value in [mav_cmd_do_mount_control](http://planner.ardupilot.com/wiki/common-mavlink-mission-command-messages-mav_cmd/#mav_cmd_do_mount_control) here. 
   
   * What values can the enum take that ardupilot supports

2. Is the description going to be exactly the same for Rover/Plane?

3. Are there any other recent changes around camera/mount that need documentation?

squilter, 2015-10-05:
@hamishwillee 

1. yes.  It will set it to any value in the enum, so nothing will get outright rejected.  But I'm not sure if the vehicles support retracting a mount.  maybe they do, not sure.
2.  yes.
3.  Only to note that `mount_control` and `mount_configure` are deprecated now.  But I didn't see those in the docs anyway.

hamishwillee, 2015-10-05:
@squilter Thank you. I have updated [MAV_CMD_DO_MOUNT_CONTROL](http://planner.ardupilot.com/wiki/common-mavlink-mission-command-messages-mav_cmd/#mav_cmd_do_mount_control) and linked it from the rover and plane sections too. Can you please "sanity" check it?

I also added links to these from the [Plane Commands in Guided Mode](http://dev.ardupilot.com/wiki/plane-commands-in-guided-mode/#mav_cmds) and [Copter Commands in Guided Mode](http://dev.ardupilot.com/wiki/copter-commands-in-guided-mode/#mav_cmds) topics.

Re "3" do mean that these commands MAV_CMD_DO_MOUNT_CONTROL and MAV_CMD_DO_MOUNT_CONFIGURE are deprecated? If so, in favour of what alternative command(s) - MAV_CMD_DO_DIGICAM_CONTROL / CONFIGURE? 

@rmackay9 Is this a "formal" deprecation - ie should I add a note against these methods in the above docs?  docs too?

hamishwillee, 2015-10-05:
@squilter PS Copying the mode parameter for MAV_CMD_DO_MOUNT_CONTROL  doesn't appear to be [in the master branch](https://github.com/diydrones/ardupilot/blob/master/libraries/AP_Mission/AP_Mission.cpp#L994) - does this get merged in later?

squilter, 2015-10-06:
There were old messages called `mount_control` and `mount_configure`.  Those should not be used any more.  They have been replaced with `MAV_CMD_DO_MOUNT_CONTROL` and `MAV_CMD_DO_MOUNT_CONFIGURE`.

hamishwillee, 2015-10-06:
Ah, thank you!