ArduPilot/ardupilot #2672: Add new camera trigger feature

jakubjon, 2015-08-07:
Hi,
1) I am totally newbie in Ardupilot stuff, but I would like to use Pixhawk as a controller for Ultra Light plane camera gimbal. I plan to use Arduino instead of Rx and I would like to trigger camera each certain distance. Is there any chance to set CAM_TRIGG_DIST using PWM input? Or in the best case to turn off and on triggering using different PWM input?

2) In case there is no such an option how would you estimate difficulty of adding such feature by my self? I would define my self as a basic C++ programmer with arduino experiences.
A was actually able to compile PX4 firmware (without loading it in to Pixhawk). But I found out that I don't understand to the connection between PX4 and APM firmwares. Just a  hint or lik in this case will help me..

Thanks Jakub

squilter, 2015-08-07:
> I would like to trigger camera each certain distance

Try [this](http://copter.ardupilot.com/wiki/common-optional-hardware/common-cameras-and-gimbals/common-camera-shutter-with-servo/)

> Is there any chance to set CAM_TRIGG_DIST using PWM input

Pixhawk doesn't support PWM input, so there isn't really a clean way to do it.  The proper way to change the distance is to send a mavlink command called `DO_SET_CAM_TRIGG_DIST`

> estimate difficulty of adding such feature by my self?

That depends on how you end up implementing it. I would recommend not modifying the ardupilot code.  Instead, use [dronekit](dronekit.io) to send `DO_SET_CAM_TRIGG_DIST` or `MAV_CMD_DO_DIGICAM_CONTROL` or something else like that.

>  Just a hint or lik in this case will help me.

dev.ardupilot.com is pretty great

closed 2015-08-07 by squilter

rmackay9, 2015-08-08:
A small correction, the Pixhawk does support PWM input, we use it for the lidar-lite driver.
http://copter.ardupilot.com/wiki/common-optional-hardware/common-rangefinder-landingpage/common-rangefinder-lidarlite/#connecting_to_pixhawk_via_pwm

I think the rest of @squilter's advice is good.  An alternative would be to create a new Ch7/Ch8 auxiliary switch function to enable/disable the camera trigger.

rmackay9, 2015-08-08:
By the way, feel free to email on drones-discuss@googlegroups.com or join gitter (see "How the team works" on this link: http://dev.ardupilot.com)

jakubjon, 2015-08-11:
Thank you for your hints. I decided to use PWM to ppm encoder to operate gimbal and Arduino to set CAM_TRIGG_DIST parameter via telemetry using MAVlink library. But after many tries I didn't figure out how to create such a message.. I have added MAVlink c library but I cant find proper command. Please may I ask you to direct me on peace of code for inspiration?
Thanks

djnugent, 2015-08-11:
@jakubjon mavlink on arduino uno does not work(anymore). It does not have enough SRAM. You will need to use a bigger arduino, like the Mega or a Teensy.