ArduPilot/ardupilot #2684: Arduplane flying between waypoints with a consistent offset

hemav, 2015-08-10:
We are flying a X5 Flying Wing [with ArduPlane V3.3.0 (6be0932d)], and sometimes (not always, but it is consistent within the flight) it will fly with an offset relative to the line defined in the mission plan. As a result, the flight has pairs of lines closer than they should, and far from the next pair of lines. It looks like a software/configuration issue, since the lines themselves are neat and straight. 

Here are the parameters used, the plan for a flight that showed the fault, and the result of that flight and another flown back-to-back, which was also faulty.

Parameters: http://pastebin.com/0NUf41Es

[image]

[image]

iskess, 2015-08-10:
This is a common problem for a lot of users, however I have never heard of someone experiencing this with a flying wing. Usually the problem is with a Skywalker (Aero). Most people resolve the issue with enormous amounts of rudder trim, to the point that it won't fly straight in Manual mode. Obviously that's not an option for you. Please read the following threads: 
http://ardupilot.com/forum/viewtopic.php?t=12757

magicrub, 2015-08-10:
Is this a duplicate of this issue? https://github.com/diydrones/ardupilot/issues/2650

hemav, 2015-08-10:
@iskess Yeah, I had read those threads and recognize you from them. But since the rudder trick can't fix this case (plus since it varies in gravity and side, the trim would need readjusting every time), I'm trying to get to the source of the issue.

@magicrub You're right, the same behaviour is observed here. But with the turns in between, it looks even more surprising that it consistently goes back to around the same value of xtrack_error. It's like if some erroneous constant factor is applied to the calculation that decides where should the plane try to go, and that factor also looks to be relative to the plane orientation, since it always seems to be towards the same side. In the pink flight it goes to ~6 meters to the right, and in the orange flight it goes ~11 meters to the left.

[image]

[image]

magicrub, 2015-08-10:
@hemav yes, this looks exactly like the same cross track problem that @WickedShell and I can duplicate. To keep things clean I'm going to close this and mark it duplicate since the other one was opened first. Let's keep track of this, it's a serious issue!

magicrub, 2015-08-10:
Duplicate of https://github.com/diydrones/ardupilot/issues/2650

closed 2015-08-10 by magicrub

tridge, 2015-08-12:
If you have a tlog then can you post it? I've just fixed a bug where a change in the mode switch position would change the prev_WP (the start of the tracking leg) even if it didn't change flight mode. I'd like to see if your tlog records a change in the flight mode channel

hemav, 2015-08-12:
@tridge 
Negative I think, channel 5 stays flat during the entire auto part. You have the tlogs on your email, for use by you and the rest of the dev team.