ArduPilot/ardupilot #2701: Copter: Ch6 tuning function broken when booting up with no RC Control

R-Lefebvre, 2015-08-16:
Currently if the user is using a Ch6 Tuning Function, then shuts down, and starts up the copter without turning on the Tx, the value being tuned will become Zero.  This could result in a crash, especially if the user is attemting an RC-less mission using a GCS.

Copter should either remember the last known Ch6 value in a parameter for use in this situation, or, at least default the Ch6 tuning value to a midpoint between low and high.

rmackay9, 2015-08-16:
Maybe what would be best is to stop tuning if there's an rc failsafe or no rc present.  There are flags for both situations.

R-Lefebvre, 2015-11-13:
Randy, just looking at this one.  I wonder if this should be an arming check?  If there is an Active Ch6 Tuning function, and the Tx isn't on, maybe we should just refuse to arm?

rmackay9, 2015-11-14:
..or maybe we just shouldn't tune with ch6 if the TX is off (maybe we do that already)?  I worry about the arming check because it would force someone to go in and change a param just because, for this flight, they have their TX off.

R-Lefebvre, 2015-11-14:
The problem is, what value do we assign to the param if the RC is off?  I guess we could just leave it to whatever the parameter is set to.  Just sometimes, that might not be right.  But we're trying to find the least worst solution.  I'd rather warn the pilot about the problem.

Currently, the parameter gets set to 0, which is really bad.

rmackay9, 2015-11-14:
yes, i think leaving it as the original parameter value makes sense.  If we've never updated it from the tx then that's what it should be by default.

WickedShell, 2015-11-14:
Out of curiosity what happens to the parameter if you go have a loss of
link/RC failsafe?

rmackay9, 2015-11-14:
@WickedShell it should just stop updating.  i'm pretty sure that's what it does anway..

WickedShell, 2015-11-14:
If the RC receiver is set up to swap to pre defined out puts the change is
still ignored? (Hopefully)

R-Lefebvre, 2015-11-14:
I don't think that's what happens, but I'll check it.

R-Lefebvre, 2015-12-29:
Issue solved with this PR:

https://github.com/diydrones/ardupilot/pull/3391

@WickedShell, I checked the code, and if your RC Tx enters a failsafe condition, the tuning param will not change.

R-Lefebvre, 2015-12-31:
PR merged so this issue is closed.

closed 2015-12-31 by R-Lefebvre