ArduPilot/ardupilot #2702: PX4-V1:Building with AC3.3RC8 failed

touma-setsuna, 2015-08-16:
Hi, some problems occurred when I tried to build AC3.3RC8 for px4-v1 hardware.
Here's the PX4 Console info:

$ make px4-v1
// BUILDROOT=/d/APM3.3rc8/ardupilot/Build.ArduCopter HAL_BOARD=HAL_BOARD_PX4 HAL
_BOARD_SUBTYPE= TOOLCHAIN=NATIVE EXTRAFLAGS=-DGIT_VERSION="d8cee508" -DNUTTX_GIT
_VERSION="7c5ef883" -DPX4_GIT_VERSION="7a620546" -DUAVCAN=1
Checking modules
make[1]: Entering directory `/d/APM3.3rc8/ardupilot/modules/PX4Firmware'
Skipping submodules. NUTTX_SRC is set to /d/APM3.3rc8/ardupilot/modules/PX4NuttX
/nuttx/
%% Configuring NuttX for px4fmu-v1
include/arch already exists but is not a symbolic link
make[2]: *** [clean_context] Error 1
make[1]: *** [/d/APM3.3rc8/ardupilot/modules/PX4Firmware/Archives/px4fmu-v1.expo
rt] Error 2
make[1]: Leaving directory `/d/APM3.3rc8/ardupilot/modules/PX4Firmware'
make: *** [/d/APM3.3rc8/ardupilot/modules/PX4Firmware/Archives/px4fmu-v1.export]
 Error 2

Please notice that everything got stuck at this line:
%% Configuring NuttX for px4fmu-v1

Then something went wrong.

It's OK to build PX4-V2 and I have done "make clean" before building px4-v1.

Is this a bug or something? Hoping to get an answer. Thanks a lot.

squilter, 2015-08-16:
It compiles fine for me, which means this is a support question.  Please bring it up here:
http://ardupilot.com/forum/viewforum.php?f=100

closed 2015-08-16 by squilter

rmackay9, 2015-08-16:
I've confirmed as well that i compiles ok for me using make and/or eclipse on windows.

touma-setsuna, 2015-08-17:
Thanks. I'll try some other ways to compile this.