ArduPilot/ardupilot #2706: issues while building arduCopter-3.3 for pixhawk using PX4toolchain

baoyuncheng, 2015-08-17:
hello guys:
I am a newer here.
Recently i built arduCopter-3.3 using PX4toolchain as the official-guide: http://dev.ardupilot.com/wiki/building-the-code/building-px4-with-make/

but i cannot get what i wanna.it's just show like this in the px4 console:
fatal: ambiguous argument 'HEAD': unknown revision or path not in the working tree
Use '--' to separate paths from revisions
and so on...

and i noticed that in 'ardupilot-Copter-3.3-->modules-->PX4Firmware & PX4NuttX & uavcan' there are empty!!
with any step wrong could make it so bad?? i need your help,sinceriously!!

rmackay9, 2015-08-17:
maybe try running this on the console.
"git submodule update"

This is a dev question so doesn't really belong on the to-do list.  Best to email drones-discuss@googlegroups.com or join Gitter.  It's listed on the dev wiki under the "How the team works" section.  http://dev.ardupilot.com/

closed 2015-08-17 by rmackay9

baoyuncheng, 2015-08-17:
I think I have done it,but got the same…and I also did ‘git submodule init’ ‘git submodule update’ ‘git init’.  any idea?

 

This is a dev question so doesn't really belong on the to-do list. Best to email drones-discuss@googlegroups.com <mailto:drones-discuss@googlegroups.com>  or join Gitter. It's listed on the dev wiki under the "How the team works" section. http://dev.ardupilot.com/