ArduPilot/ardupilot #2716: Plane: rcin throttle jitter results in large throttle bumps

KIrill-ka, 2015-08-19:
That happens in FBWB mode, should be the same in CRUISE and AUTO with throttle nudge.
Input jitter is 1us. Output jitter is quite significant ~70us peak to peak which is almost 10% of TECS' throttle range.

After some digging into the problem I found two causes.
1. 1us jitter may result 1% jitter because the throttle is converted to 0-100 integer range.
2. TECS  reacts quite aggressively to a minor speed change fed to update_pitch_throttle.

Configuration info:
ArduPlane 3.3.0, PX4, s.bus connection to RX,
airspeed sensor enabled

Naterater, 2019-01-07:
@KIrill-ka, is this still an issue with newer firmware such as Plane 3.9? I have not witnessed the same behavior recently.

WickedShell, 2019-01-07:
@Naterater yes, it's due to internally quantizing all throttle on plane from 0-100 in steps of 1. This needs to change over time, but it's a large/slow change with a lot of implicit assumptions that might be missed.

KIrill-ka, 2019-01-07:
I'm sorry, I haven't tried the recent ardupilot versions. And I can't check it now since it's too cold for me to fly.

SlipstreamFPV, 2021-03-27:
What can be done to address this issue? It would seem RC filtering is a must particularly in auto modes.

It would also seem that the TECS controller is far too aggressive in chasing controller adjustments where it could have a slower response to controller changes while maintaining a fast response to external factors as required.

magicrub, 2021-03-27:
I actually have a fix for this on my fork but haven't brought it in yet. It's part of a zillion other changes that I need to get merged

SlipstreamFPV, 2021-03-27:
That's great news! I'm really looking forward to it being merged as I've struggled with this on a few planes. Thanks for all your hard work.

IamPete1, 2022-09-05:
The 100 steps on the output thing has been fixed by https://github.com/ArduPilot/ardupilot/pull/18706. We do have a similar issue on the input, https://github.com/ArduPilot/ardupilot/pull/18985