ArduPilot/ardupilot #2717: please add support for the RPLIDAR

geofrancis, 2015-08-19:
would you be able to add support for the RPLIDAR it would be excellent for avoidance giving 360 degrees lidar coverage would be better than multiple sonar units and more reliable,

magicrub, 2015-08-19:
Looks like this is the device:
http://www.seeedstudio.com/depot/RPLIDAR-360-degree-Laser-Scanner-Development-Kit-p-1823.html

This will be generating a lot of data, via Serial port only, which is much better handled on a companion computer. @geofrancis , how do you imagine this being used in a flight controller without a companion computer?

geofrancis, 2015-08-19:
I'm not sure but I wonder if the sample rate could be reduced to a point where it could be handled by a flight controller. but still have enough resolution to be useful.

rmackay9, 2015-08-20:
It's certainly technically possible to add support and object avoidance is on the to-do list for some time within the next year and we will need multiple fixed range finders or a single scanning range finder for that project.  I don't know if this particular sensor will be good enough (for copters the 6m range is a bit short for outdoor use at least).
I don't think we know yet if we can handle multiple sensors with a Pixhawk or whether we will need a companion computer of linux autopilot (like the NAVIO+).

SteveJos, 2017-07-10:
Hello, I started the integration of the latest RPLIDAR-A2 with 16m range in the AP_PROXIMITY library. The serial communication is working and I already get measurements from the LIDAR in the proximity window, but further testing is needed . Please have a look at the PR #6552:

https://github.com/ArduPilot/ardupilot/pull/6552

rmackay9, 2018-11-14:
This is included in Copter-3.6 and Rover-3.5 so closing. Wiki describing the setup is [here](http://ardupilot.org/copter/docs/common-rplidar-a2.html).

closed 2018-11-14 by rmackay9