ArduPilot/ardupilot #2718: EKF: calculate wind.z

magicrub, 2015-08-19:
Current EKF does not calculate the z axis and is noted as so in the source:
    wind.x = state.wind_vel.x;
    wind.y = state.wind_vel.y;
    wind.z = 0.0f; // currently don't estimate this

Adding this would help pave a path to more advanced auto-path planning improvements, namely gliders auto-thermal hunting.

magicrub, 2015-08-19:
Of note, looks like DCM *does* populate the z axis so the EKF is a regression there.

priseborough, 2015-08-19:
In practice this was not a regression. The Z component is very poorly observable using the DCM method given it assumes zero angle of attack and produces offsets that make it unsuitable for path planning, thermal hunting, etc.

To do it properly would require adding some vehicle flight dynamics properties into the estimator, so that the relationship between angle of attack and lift is included. This is best done outside the EKF as it is platform specific.

-Paul

magicrub, 2015-08-19:
awwww, what a let down. So what you're saying is it's not implemented yet because it's harder than I think it is to implement it? Dang!

magicrub, 2016-01-05:
Will not implement, no benefit

closed 2016-01-05 by magicrub