ArduPilot/ardupilot #2729: Copter: log GPS and Baro glitches

rmackay9, 2015-08-21:
Since moving to the EKF in AC3.3 we've lost the logging of barometer and GPS glitches.  We should get these back because they help with support if they appear as Errors in the logs so that the support person can easily see that the GPS or Baro quality was bad.

peterbarker, 2017-12-19:
We do long GPS glitches again now.

peterbarker, 2021-07-30:
@rmackay9 how did baro glitch detection work?

rmackay9, 2021-07-30:
@peterbarker,

Inertial nav had a gate within which the baro sensor values would need to fall in order to be fused.  After a certain amount of time a reset to the baro's latest value would happen.  The EKF does the same thing I think.  The glitch and the eventual reset are a bit like bookends to these events and we should be logging the resets as well... maybe it would be sufficient just to log the resets.