ArduPilot/ardupilot #2733: GPS lat/lon == (0,0) handling

magicrub, 2015-08-21:
There are several checks in the source that change behavior if GPS lat == 0 and/or lon == 0. Due to IMU mixing and dead reckoning the GPS can drift around at zero. We need to investigate if changing all of those checks are better abstracted to a helper function with logic with something like this:

is "zero" when: gps.stats < 2d lock && (abs(gps.lat) < 1E7) && (abs(gps.lon) < 1E7)

rmackay9, 2015-08-22:
I've marked these as Plane issues but feel free to correct if it's for all vehicles.

magicrub, 2015-08-23:
Yeah, it is project wide. There no specific example that i can think of, just seemed like it was a gremlin that is lurking that could be causing thinks but other logic of catching... For now.

WickedShell, 2015-08-23:
Where do we look at gps lat/lon being 0 with special behavior? Since those are technically valid coordinates I suspect it would make more sense to look at the GPS driver status rather then lat/lon (although I obviously haven't looked at any instances of this yet).

magicrub, 2015-08-23:
To validate this issue, I need to provide since examples. Until then this
is just me recalling seeing the code as i was working on some other feature
and it stuck in my head as a to-do fix for later.

Its on me to provide some examples.

magicrub, 2015-08-24:
Turns out that I only saw this in mission commands where the lat/lon was checking for zero. Closing..

closed 2015-08-24 by magicrub