ArduPilot/ardupilot #2734: V3.2.1 (36b405fb) does not initialize

Letme, 2015-08-22:
The above version which is marked as stable does not initialize on HobbyKing Micro APM 2.72 . Once you put on plane firmware everything works and calibrates flawlessly, but with the above Copter version it does not finish initialization although I could read and change parameters. This was also noticable that all arguments on Primary Flight Display were frozen and no other updates were coming.

Commit 36b405fb is broken.

magicrub, 2015-08-22:
Works on official APM 2.6 hardware which there are several tens of
thousands of happy customers. Sounds like a hobby king problem.

Letme, 2015-08-22:
We could try to troubleshoot? I am now flying on beta version just fine...

rmackay9, 2015-08-22:
It could be many things but one guess is that the boot loader that's been put on the board is too large and hasn't left enough flash space for the main program.
As Tom says though, this seems to be an individual board issue and besides this issues list is not for support items.

closed 2015-08-22 by rmackay9