ArduPilot/ardupilot #2750: Plane: is_flying vibration check to allow GPS ground movement to override it

magicrub, 2015-08-25:
is_flying to use GPS with vibration check. If the vibes say we're still, this (used to) set is_flying to false. However, if there's a good GPS lock and there's movement then override it and say false.

This is mostly a SITL workaround but if for some reason someone has a *very* vibrationally isolated aircraft then at least check it against GPS before we trigger anything funny which may drop you to of the sky via crash detection.

OXINARF, 2017-06-04:
Is this still valid? There are GPS checks but I'm not sure if it fulfills the request in this issue.

magicrub, 2017-06-04:
Still needs to be done. There are lots of corner cases I'd like to add to is flying to make it better