ArduPilot/ardupilot #2753: Reporting Current via Mavlink when Copter is disarmed

Clooney82, 2015-08-25:
I saw that Ardu:Copter isn´t reporting current over mavlink until it is armed.
I have a some additional HW (LEDs, flight recorder, ...) on my copter, and it find it useful it current is also reported over Mavlink when copter is disarmed.

thanks

KiwiHC16, 2015-08-29:
I just did a SITL test and Voltage, Current and remaining Batt are reported armed or disarmed.
Could you describe the configuration and how you observe this behavior ?
I'm going to try with a real HW (just need to find a way to test it).

Clooney82, 2015-08-29:
I working with this project: https://github.com/Clooney82/MavLink_FrSkySPort/blob/s-c-l-v-rc-opentx2.1
and I´m ready Mavlink Message No.1 with this command: ap_current_battery = mavlink_msg_sys_status_get_current_battery(&msg); 
And I only receive values when Copter is armed.
I´m using a pixhawk with AC3.3rc10.

KiwiHC16, 2015-08-29:
Looks like your link doesn't work. I got a 404 error message from the Git server.
First test with real HW shows that Voltage, Current and remaining are reported armed or disarmed. Disarmed, the value for current is 0A. I don't know if it due to a very low current or due to a SW issue. More investigation needed. 
"I´m ready Mavlink Message No.1 with this command:": Where are you reading ? In the copter ? in a GCS receiving the mavlink message ?

Clooney82, 2015-08-30:
sorry for wrong link, correct one: https://github.com/Clooney82/MavLink_FrSkySPort/tree/s-c-l-v-rc-opentx2.1
will test again next week, when I´m back from holidays.

Clooney82, 2015-09-10:
Yesterday I had the chance to test again with AC3.3-rc10 looks good, maybe I had problem in earlier AC3.3-rc.
Also today with rc11 it work.
From my side issue can be closed.

rmackay9, 2015-09-11:
great, thanks for confirming.

closed 2015-09-11 by rmackay9