ArduPilot/ardupilot #2755: Plane: Auto take off bug, LIDAR Lite

Jman841, 2015-08-25:
We are seeing a strange bug on Arduplane 3.3, I think it is related to the LIDAR Lite as I have never seen this issue before using the Lidar on this plane.

If we set the plane on the ground so the LIDAR is blocked (minimum value to accept readings is 50cm), switch it to Auto mode to ensure the elevator moves in the correct direction, then, switch back to manual. Pick the plane up above our head, and switch the plane to Auto, the elevator response is no longer correct. Instead of the elevator going up, it goes to a stabilize mode like it thinks the plane is flying. This happens even if we have not yet armed the motors and is repeatable.

magicrub, 2015-08-25:
If I read that right, you said you set it in AUTO, then MANUAL, then raise
the aircraft up and when switching to AUTO it actually goes into STABILIZED?

Was it currently executing a TAKEOFF mission item?

Jman841, 2015-08-25:
That is correct.

Sometimes it works fine, but, sometimes it acts like it is already flying or at altitude. I'll try and get a video of it or a log file.

The vehicle is in Auto mode, but, the elevator does not function as it should for a takeoff waypoint. and yes, the first waypoint is take off with 15 degree's of pitch and altitude of 150 meters.

Jman841, 2015-08-25:
Of course, I can't get it to do it today, however, even though the throttle is disarmed the GCS reported 20 seconds of "time in air"

tridge, 2015-08-25:
a video isn't useful. A log file is what is needed to work out what is going on. A tlog is fine

Jman841, 2015-08-25:
I found a .tlog file that represents the issue. If you notice it is in auto with waypoint 1 set (it is the takeoff command) and the elevator output is at neutral when the plane is level.

This is a setup that has been flown multiple times and only started seeing these weird issues since installing the LIDAR sensor.

Here's the .tlog start it at 25%: https://www.dropbox.com/s/6fiocc45dtfrg1m/2015-08-22%2019-52-59.tlog?dl=0

I think it may also be related to this issue as I still have not figured out what happened here: http://ardupilot.com/forum/viewtopic.php?f=110&t=13441

Naterater, 2019-01-07:
I think we've had enough success with LiDAR Lites recently that this issue has probably been solved since then. @Jman841, can you reproduce this on the newer firmwares (3.9)?

closed 2021-07-26 by Naterater