ArduPilot/ardupilot #2763: RTL_AUTOLAND=2 as default?

magicrub, 2015-08-27:
I propose changing RTL_AUTOLAND default from 0 to 1.
from http://plane.ardupilot.com/wiki/arduplane-parameters/#rtl_auto_land_arduplanertl_autoland

RTL auto land (ArduPlane:RTL_AUTOLAND)
Automatically begin landing sequence after arriving at RTL location. This requires the addition of a DO_LAND_START mission item, which acts as a marker for the start of a landing sequence. The closest landing sequence will be chosen to the current location.

VALUE	MEANING
0	Disable
1	Enable – go HOME then land
2	Enable – go directly to landing sequence

proficnc, 2015-08-27:
Love it!

magicrub, 2015-08-27:
Whoops, meant change from 0 to 2

WickedShell, 2015-08-27:
What did the GCS show? Ideally it's showing RTL plus correct weep numbers,
otherwise I think the behavior would be alarming and look like a fly away.

Am I correct in assuming that rtl airplane of 1 or 2 without a landing
sequence will behave the same as the behavior of 0?

magicrub, 2015-08-27:
That's right, @WickedShell . If there's no DO_LAND_START then it jumps to index 0 so that's the same behavior as 0 which is to loiter at home forever. It's quite silly. It should be that if you've added a DO_LAND_START then it just works, shouldn't have to then enable it with RTL_AUTOLAND >0. There shouldn't even be a disable (0) option for RTL_AUTOLAND. if you don't want it to jump to DO_LAND_START then don't put them in your mission!

tridge, 2015-09-24:
I prefer to have an option for this. If you setup your mission somewhere else and then fly at a new location, then I don't want it flying to my old location when I RTL!

closed 2015-09-24 by tridge