ArduPilot/ardupilot #2766: Plane: Enhancement: Throttle Nudge on Approach to Land

iskess, 2015-08-28:
It is possible that an airplane with a wide speed range is using TECS_LAND_ARSPD to fly at an approach speed much closer to stall (1.3 Vso) . In the case of wind shear on approach or other factors, the operator might want to nudge the throttle to increase speed to protect the stall margin. This also gives the operator the flexibility to delay the decrease in target speed until on short final, like after crossing the tree line for example. However, I don't think it should be possible to nudge past TRIM_ARSPD_CM while on approach. So it should be designed that during approach, the bottom half of the throttle stick range becomes active for nudge, rather than the top half in other phases of flight. I believe people will be more apt to use the TECS_LAND_ARSPD option if they control it with their throttle stick. 
I propose the throttle stick movement follows the following schedule during the approach phase.
0% to 50% = Nudges airspeed proportionally between TECS_LAND_ARSPD and TRIM_ARSPD_CM
51% to 94% = TRIM_ARSPD_CM
95% to 100% Triggers the landing abort according to the proposed LAND_ABORT_THR (#2759) 
.

magicrub, 2015-08-28:
PR 2759 includes a new param "LAND_ABORT_THR ". How does that come into
play? Maybe we can make a new param that this and PR2759 can share since
they're all land throttle related?

magicrub, 2016-05-19:
is this still wanted?

iskess, 2016-05-19:
I still think it would be a good enhancement that gives the operator some airspeed control to handle dynamic situations, but still offering protections. I don't see any reason not to add this.

Naterater, 2019-01-04:
I'm not a big fan of the airspeed nudge logic on land because typically TECS will command pitch-down to nudge airspeed. That can be very dangerous on landing. @iskess, do you still want this feature?

IamPete1, 2021-09-12:
@iskess any response to the above?